project-chip/connectedhomeip #27636: Unable to build pregen files

phonnakasturi-apple, 2023-07-05:
### Reproduction steps

1. cd ~/connectedhomeip
2. export ZAP_INSTALL_PATH=~/Downloads/zap-mac/
3. scripts/codepregen.py ${OUTPUT_DIRECTORY:-./zzz_pregenerated/}
4. export PKG_CONFIG_PATH="/opt/homebrew/opt/openssl@3/lib/pkgconfig"
5. source scripts/bootstrap.sh
6. source scripts/activate.sh
7. cd ~/tools
8. git clone https://github.com/espressif/esp-idf.git
9. cd esp-idf
10. git pull
11. git checkout v4.4.4
12. git submodule update --init
13. ./install.sh
14. . ./export.sh
15. cd ~/connectedhomeip/examples/all-clusters-app/esp32
16. export ZAP_INSTALL_PATH=~/Downloads/zap-mac/
17. idf.py -DCHIP_CODEGEN_PREGEN_DIR=~/connectedhomeip/zzz_pregenerated/ build

### Bug prevalence

always

### GitHub hash of the SDK that was being used

fcf37c239c5c9858d312030a4549dbe91cee8e2b

### Platform

darwin

bzbarsky-apple, 2023-07-05:
@shubhamdp @andy31415 @dhrishi

andy31415, 2023-07-05:
Did you try to use `build_examples.py` since this is what our CI is using?

https://github.com/project-chip/connectedhomeip/blob/master/.github/workflows/examples-esp32.yaml#L78

I am surprised things do not compile since CI is supposed to validate it, however at the same CI runs on linux and using build_examples, so maybe some differences there.

bzbarsky-apple, 2023-07-05:
> Did you try to use build_examples.py since this is what our CI is using?

That is not viable since it does not let you flash/monitor....

andy31415, 2023-07-06:
> > Did you try to use build_examples.py since this is what our CI is using?
> 
> That is not viable since it does not let you flash/monitor....

I am thinking that build_examples.py also just runs shell commands for building, so if build_examples.py works then we probably have set of commands that succeeds compiling.

I see:

```shell
# Commands will be run in CHIP project root.
cd "."

# Generating esp32-m5stack-all-clusters
mkdir -p /workspace/out/esp32-m5stack-all-clusters

cp examples/all-clusters-app/esp32/sdkconfig_m5stack.defaults /workspace/out/esp32-m5stack-all-clusters/sdkconfig.defaults

rm -f examples/all-clusters-app/esp32/sdkconfig

bash -c 'source $IDF_PATH/export.sh; source scripts/activate.sh;
export SDKCONFIG_DEFAULTS=/workspace/out/esp32-m5stack-all-clusters/sdkconfig.defaults
idf.py -C examples/all-clusters-app/esp32 -B /workspace/out/esp32-m5stack-all-clusters -DCHIP_CODEGEN_PREGEN_DIR=/workspace/zzz_pregenerated reconfigure'

rm -f examples/all-clusters-app/esp32/sdkconfig

# Building esp32-m5stack-all-clusters
bash -c 'source $IDF_PATH/export.sh; source scripts/activate.sh;
export SDKCONFIG_DEFAULTS=/workspace/out/esp32-m5stack-all-clusters/sdkconfig.defaults
idf.py -C examples/all-clusters-app/esp32 -B /workspace/out/esp32-m5stack-all-clusters build'
```

so it looks like it does a `-B` first to generate, but then compilation is done without setting that option. I wonder if that is the difference.

andy31415, 2023-07-06:
Tried in a vscode environment via:

```sh
cd examples/all-clusters-app/esp32
. /opt/espressif/esp-idf/export.sh
idf.py -DCHIP_CODEGEN_PREGEN_DIR=/workspace/zzz_pregenerated/ build
```

and it compiled, so I wonder if the issue is mac specific after all. Although #27654 proposes a fix.

andy31415, 2023-07-06:
could you share the steps for Ubuntu failure? I am still failing to reproduce on the `chip-build-vscode` image but unsure what steps are relevant and what not  ... I may be running things out of order somehow.

andy31415, 2023-07-06:
Got it reproduced (thanks @bzbarsky-apple !) - need to `rm sdkconfig` first.

andy31415, 2023-07-06:
When I removed sdkconfig, I got `CONFIG_DEVICE_TYPE_ESP32_DEVKITC=y` so I am guessing it just pulls in some defaults.

I did
```
rm sdkconfig
cp sdkconfig_m5stack.defaults sdkconfig
```

and the build seemed fine.

andy31415, 2023-07-06:
testing, when things fail I see it tries to run:

```
/home/vscode/pigweed/env/cipd/packages/pigweed/gn --root=/workspace/config/esp32 gen --check --fail-on-unused-args /workspace/examples/all-clusters-app/esp32/build/esp-idf/chip
```

and this is identical when it succeeds.
The only diff in args.gn seems to be a `chip_build_tests=true`

andy31415, 2023-07-06:
When in a `fail status` I get:

```
gn --root=/workspace/config/esp32 gen --check --fail-on-unused-args /workspace/examples/all-clusters-app/esp32/build/esp-idf/chip
ERROR at build arg file (use "gn args <out_dir>" to edit):14:37: Build argument has no effect.
chip_code_pre_generated_directory = "/workspace/zzz_pregenerated/"
```

so it is reproducible just running GN. What I am unclear about is what happens when it works. It seems that IDF changes some things internally in `config/esp32`

andy31415, 2023-07-06:
The diff seems to be just `chip_build_tests=true`.
Compile succeeds with that set, fails without it.

Apparently sdkconfig defaults enable build tests and I am guessing tests depend on some pregen bits, but general library compile for esp32 does not depend on pregen.

andy31415, 2023-07-06:
@phonnakasturi-apple  workarounds:
- ensure `CONFIG_BUILD_CHIP_TESTS` is set to y (if you copy one of the sdkconfig.defaults files this will be the case)
- edit `config/esp32/components/chip/CMakeLists.txt` and remove `--fail-on-unused-args`

shubhamdp, 2023-07-06:
> ```
> rm sdkconfig
> cp sdkconfig_m5stack.defaults sdkconfig
> ```
> 
> and the build seemed fine.

`sdkconfig` is autogenerated, when we do a fresh build there will be default configurations generated out of Kconfig files, but we may want to override some or it and we do that using `sdkconfig.defaults` file. Options in `sdkconfig.defaults` is like a patch on top of them.

Above change did not do anything just built again using the default configurations.

I did disable `CONFIG_BUILD_CHIP_TESTS` by default as it builds tests and I though they are unnecessary in examples to reduce CI time. (So... regression?)

I even tried without `--fail-on-unused-args` but I though it may not be a good fix. If this sounds good, I can make `--fail-on-unused-args` conditional if `CHIP_CODEGEN_PREGEN_DIR` is set.

PR #27654 (2023-07-08): [ESP32] Fix the build with pregenerated files
#### Problem
#27636

#### Change Overview
Fixes #27636
Do not use --fail-on-unused-args flag when using pregenerated file
#### Tests
```
cd connectedhomeip/examples/all-clusters-app/esp32
# Successfully built example using both methods
idf.py -DCHIP_CODEGEN_PREGEN_DIR=/Users/shubhampatil/esp/connectedhomeip/zzz_pregen build
idf.py build
```
closed 2023-07-08 by mergify[bot]