project-chip/connectedhomeip #27650: [BUG] [TC-ACT-2.2] INVALID_ACTION returned when subscribing ActionFailed event

Survensa, 2023-07-06:
### Reproduction steps

1. Bring up the DUT with bridge-app.
2. Pair the DUT using chip-tool through "onnetwork" transport.
3. Run the command `./chip-tool interactive start` and `actions subscribe-event action-failed 2 30 1 0`

**Expected Outcome:**

> [1659962630.456475][25381:25386] CHIP:DMG: SubscribeResponse is received
> [1659962630.456497][25381:25386] CHIP:DMG: SubscribeResponseMessage =
> [1659962630.456509][25381:25386] CHIP:DMG: {
> [1659962630.456520][25381:25386] CHIP:DMG:         SubscriptionId = 0x78f0a04c,
> [1659962630.456531][25381:25386] CHIP:DMG:         MaxInterval = 0x64,
> [1659962630.456542][25381:25386] CHIP:DMG:         InteractionModelRevision = 1
> [1659962630.456551][25381:25386] CHIP:DMG: }

**Actual Outcome:**

> [1688633445.057159][4764:4766] CHIP:DMG: StatusResponseMessage =
> [1688633445.057225][4764:4766] CHIP:DMG: {
> [1688633445.057282][4764:4766] CHIP:DMG: 	Status = 0x80 (INVALID_ACTION),
> [1688633445.057345][4764:4766] CHIP:DMG: 	InteractionModelRevision = 1
> [1688633445.057401][4764:4766] CHIP:DMG: }
> [1688633445.057490][4764:4766] CHIP:IM: Received status response, status is 0x80 (INVALID_ACTION)
> [1688633445.057726][4764:4766] CHIP:EM: <<< [E:17011i S:49670 M:99740487 (Ack:51058832)] (S) Msg TX to 1:0000000000000001 [AFDF] --- Type 0001:01 (IM:StatusResponse)

### Bug prevalence

Whenever I try to Subscribe ActionFailed event

### GitHub hash of the SDK that was being used

7ae5919a046a18c7e60537aff814977ee424b9cf - Master

### Platform

raspi

### Platform Version(s)

RPI - 4, 8GB RAM

### Anything else?

While subscribe ActionFailed event in this commit 7806dd19111513a61bb90d70a0e392dc89eeb454, the subscription is completed successfully with SubscriptionId.

But while validating with the commit 7ae5919a046a18c7e60537aff814977ee424b9cf, It returned as INVALID_ACTION.

bzbarsky-apple, 2023-07-06:
Presumably the DUT's config claims it does not support this event.

cjandhyala, 2023-07-06:
@Survensa is this a regression ? In the zap file I don't see this event enabled.

bzbarsky-apple, 2023-07-08:
The change to respond INVALID_ACTION is due to a spec alignment.  Before this we silently (and incorrectly per spec) created the subscription even though it did not have any valid paths (due to the incorrect ZAP config).

cjandhyala, 2023-07-10:
Thx @bzbarsky-apple , So this is WAD(Works as Designed), we"ll close the issue.

bzbarsky-apple, 2023-07-11:
Well, there is an issue here: the example bridge-app has an incorrect ZAP config....

closed 2023-10-12 by Survensa

Survensa, 2023-10-12:
The endpoint for Actions cluster has been changed from 0 to 1. So If we subscribe this event using endpoint 1, the subscription will be completed successfully. So Closing this issue