project-chip/connectedhomeip #27651: [Feature] Multiple Bridges in a single service

mustafayuce33, 2023-07-06:
### Feature description

I am just wondering if multiple matter bridges can be published in a single application? In an other saying can we create multiple fabrics and manage it in a single App and be added in Apple Home? This means that I will be seeing multiple Avahi broadcast in one bridge app. 

### Platform

raspi

bzbarsky-apple, 2023-07-06:
@mustafayuce33 A single Matter bridge can bridge to different backing network technologies.  For example, you could have a device that looks like a single Matter node but exposes both Zigbee and Z-Wave devices, or something like that.

Does that answer your question?  If not, can you explain more clearly what your question is, please?

mustafayuce33, 2023-07-06:
Thanks for your quick reply. What I mean is that I want to see multiple bridge in Apple Home. But all these bridges wants to be managed in single App. Think out I have build bridge app and run it as follows;

```
cd ~/connectedhomeip/examples/bridge-app/linux
sudo out/debug/chip-bridge-app --ble-device [bluetooth device number]
```
What I want this app to publish multiple bridges (lets say 10). So that I can assign all these bridges in different homes in Apple home. 

Think about select accessory page in apple home shown below;

[image]

I want to see multiple bridges published by single matter app with names lets say "Matter Bridge 1", "Matter Bridge 2" and so on.

I can achieve it by running multiple bridge app with command 

$ ./bridge-app --passcode="97521240" --discriminator="3245"
$ ./bridge-app --passcode="67901240" --discriminator="4567"
$ ./bridge-app --passcode="12901240" --discriminator="8767"
.......

But I do not want that kind of solution. I want to send a parameter to bridge app and it will publish multiple bridge. I hope I have explained it clearly.

bzbarsky-apple, 2023-07-08:
Yes, that's clear.  That does not exist right now, and you would have to build something like that if you want it.

It's rather difficult, because it's hard-to-impossible to run multiple Matter stacks in a single process right now.

closed 2023-09-04 by mustafayuce33