project-chip/connectedhomeip #27652: Is it possible on Apple to bind the light bulb to the switch through a tool like chip-tool

chuckrui, 2023-07-06:
Hi~
I used Raspberry as a Board Router, then created a Thread network, 
and added nordic routines light_bulb and light_switch to the Thread network. 
I then bind the light_bulb to the light_switch using the chip-tool implementation. 
I was wondering if I wanted to add the light_bulb and light_switch to my home app and implement it using the chip-tool, what should I do? 
Or is there another way, in an Apple-based ecosystem, to do the same thing by binding light_bulb to light_swtich

bzbarsky-apple, 2023-07-06:
I'm not sure what you mean by "implement it using the chip-tool", but:

1. If you are commissioning devices into Apple Home, but want to set up Matter bindings between them, Home itself does not expose that (though it does have automations).  You may be able to talk to Home and ask it to set up bindings for you from an iOS app, but that's not a core Matter issue.  Please file a feedback assistant ticket about this.
2. If you are commissioning devices into your own fabric, you can set up bindings using Matter.framework, just like you can write any other attributes.

closed 2023-07-06 by bzbarsky-apple

chuckrui, 2023-07-07:
Let me restate, we all know that nordic light_switch is currently displayed as an unsupported accessory when added to the home app, not an uncertified device, while light_bulb added to the home app is displayed as an uncertified device. light_bulb can be adjusted on/off and levelcontrol in the home app. If the light_switch displays as an unauthenticated device when added to the home app, I would like to know how the light_switch is bound to the switch in the home app. As you said, Apple does not disclose a similar chip-tool tool, because I built a Thread network, I can easily use chip-tool to bind light_bulb to light_switch. You can control the on/off and levelcontrol of the light_bulb using the light_switch.

bzbarsky-apple, 2023-07-08:
>  would like to know how to bind the light bulb example to the switch in the home app

Right now I don't believe you can.  But you should file a feedback assistant ticket to check for sure.

chuckrui, 2023-07-10:
> But you should file a feedback assistant ticket to check for sure.
Do you mean I submit a new issue here？
[https://github.com/project-chip/connectedhomeip/issues/new/choose](url)
Feature Request or Test Issue ？ or build Issue？
Thank you~

bzbarsky-apple, 2023-07-10:
@chuckrui No, I mean you should file a ticket at https://developer.apple.com/bug-reporting/#feedback-assistant, probably using https://feedbackassistant.apple.com/new-form-response

The link you have there is for issues in the open-source SDK.  Your question is about an Apple product, not the open-source SDK.

chuckrui, 2023-07-11:
ok，Thank you~