project-chip/connectedhomeip #27653: Facing Issue with building lightning-app example for esp32s3 Module [1.1]

Darshanr5051, 2023-07-06:
### Reproduction steps

cd connectedhomeip/examples/lightning-app/esp32

Build is Stuck at "Solving dependencies requirements "

Also mentioning the logs.

### Bug prevalence

Whenever i attempt to build my esp32s3 as a lightning-app

### GitHub hash of the SDK that was being used

connectedhomeip

### Platform

esp32

### Type

Core SDK Performance Improvement

### Anything else?

idf.py build 

/home/User/Desktop/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. 
See https://setuptools.pypa.io/en/latest/pkg_resources.html                          
import pkg_resources                                                                                                                                                                                             Executing action: all (aliases: build)                                                                                                                                                                             Running cmake in directory /home/darshan/Desktop/connectedhomeip/examples/lighting-app/esp32/build                                                                                                                 Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/darshan/Desktop/connectedhomeip/examples/lighting-app/esp32"...                                                         
-- IDF_TARGET not set, using default target: esp32                                                                                                                                                                 -- Found Git: /usr/bin/git (found version "2.34.1")                                                                                                                                                                -- The C compiler identification is GNU 8.4.0                                                                                                                                                                      -- The CXX compiler identification is GNU 8.4.0                                                                                                                                                                    -- The ASM compiler identification is GNU                                                                                                                                                                          -- Found assembler: /home/darshan/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc                                                                                     
-- Detecting C compiler ABI info                                                                                                                                                                                   -- Detecting C compiler ABI info - done                                                                                                                                                                            -- Check for working C compiler: /home/darshan/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped                                                              
-- Detecting C compile features                                                                                                                                                                                    -- Detecting C compile features - done                                                                                                                                                                             -- Detecting CXX compiler ABI info                                                                                                                                                                                 -- Detecting CXX compiler ABI info - done                                                                                                                                                                          -- Check for working CXX compiler: /home/darshan/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped                                                            
-- Detecting CXX compile features                                                                                                                                                                                  -- Detecting CXX compile features - done                                                                                                                                                                           -- Building ESP-IDF components for target esp32                                                                                                                                                                    Skipping optional dependency: espressif/mdns                                                                                                                                                                       Solving dependencies requirements                                                                                                                                                                                  .....

bzbarsky-apple, 2023-07-06:
@shubhamdp @dhrishi

PSONALl, 2023-07-14:
@Darshanr5051 Please remove the cache for the Espressif ComponentManager. For example:

* On Linux: `rm -rf ~/.cache/Espressif/ComponentManager/`
* On macOS: `rm -rf ~/Library/Caches/Espressif/ComponentManager`

shubhamdp, 2023-08-10:
@Darshanr5051 Please close the issue if resolved

closed 2024-10-25 by dhrishi