project-chip/connectedhomeip #27655: [BUG] [TC-BRBINFO-1.1] EventList has to be updated as per SDK implementation

Survensa, 2023-07-06:
### Reproduction steps

1. Bring up the DUT with bridge-app.
2. Pair the DUT using chip-tool through "onnetwork" transport.
3. Run the command `./chip-tool bridgeddevicebasicinformation read event-list 1 3`

### Bug prevalence

Whenever I try to read EventList.

### GitHub hash of the SDK that was being used

7ae5919a046a18c7e60537aff814977ee424b9cf - Master

### Platform

raspi

### Platform Version(s)

RPI - 4, 8GB RAM

### Anything else?

As per spec the mandatory Event ReachableChanged is implemented in SDK. But in EventList attribute it's not listed.

[image]

bzbarsky-apple, 2023-07-06:
Sounds like the ZAP config for this example app needs to be fixed.

cjandhyala, 2023-08-09:
since event list is descoped from matter, closing the issue. Feel free to re-open @Survensa

closed 2023-08-09 by cjandhyala