project-chip/connectedhomeip #27668: [Build] Darwin build failure

woody-apple, 2023-07-06:
### Build issue(s)

https://github.com/project-chip/connectedhomeip/actions/runs/5472582066/jobs/9964942866
```
[817/844] touch mac_x64_gcc/obj/scripts/py_matter_yamltests/matter_yamltests.tests.test_pics_checker.py.stamp
[818/844] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../scripts/py_matter_yamltests --default-toolchain=//third_party/pigweed/repo/pw_toolchain/default:default --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/scripts/py_matter_yamltests/matter_yamltests.tests.test_pseudo_clusters.py.pw_pystamp --capture-output --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/scripts/py_matter_yamltests/matter_yamltests.tests.test_pseudo_clusters.py_metadata_path_list.txt -- ../../scripts/py_matter_yamltests/test_pseudo_clusters.py
[819/844] touch python/obj/scripts/py_matter_yamltests/matter_yamltests.tests.test_pseudo_clusters.py.stamp
[820/844] touch mac_x64_gcc/obj/scripts/py_matter_yamltests/matter_yamltests.tests.test_pseudo_clusters.py.stamp
[821/844] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../scripts/py_matter_yamltests --default-toolchain=//third_party/pigweed/repo/pw_toolchain/default:default --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/scripts/py_matter_yamltests/matter_yamltests.tests.test_parser_builder.py.pw_pystamp --capture-output --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/scripts/py_matter_yamltests/matter_yamltests.tests.test_parser_builder.py_metadata_path_list.txt -- ../../scripts/py_matter_yamltests/test_parser_builder.py
[822/844] touch python/obj/scripts/py_matter_yamltests/matter_yamltests.tests.test_parser_builder.py.stamp
[823/844] touch mac_x64_gcc/obj/scripts/py_matter_yamltests/matter_yamltests.tests.test_parser_builder.py.stamp
[824/844] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../scripts/py_matter_yamltests --default-toolchain=//third_party/pigweed/repo/pw_toolchain/default:default --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/scripts/py_matter_yamltests/matter_yamltests.tests.test_spec_definitions.py.pw_pystamp --capture-output --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/scripts/py_matter_yamltests/matter_yamltests.tests.test_spec_definitions.py_metadata_path_list.txt -- ../../scripts/py_matter_yamltests/test_spec_definitions.py
[825/844] touch python/obj/scripts/py_matter_yamltests/matter_yamltests.tests.test_spec_definitions.py.stamp
[826/844] touch mac_x64_gcc/obj/scripts/py_matter_yamltests/matter_yamltests.tests.test_spec_definitions.py.stamp
[827/844] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../scripts/py_matter_yamltests --default-toolchain=//third_party/pigweed/repo/pw_toolchain/default:default --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/scripts/py_matter_yamltests/matter_yamltests.tests.test_yaml_loader.py.pw_pystamp --capture-output --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files python/gen/scripts/py_matter_yamltests/matter_yamltests.tests.test_yaml_loader.py_metadata_path_list.txt -- ../../scripts/py_matter_yamltests/test_yaml_loader.py
[828/844] touch python/obj/scripts/py_matter_yamltests/matter_yamltests.tests.test_yaml_loader.py.stamp
[829/844] touch mac_x64_gcc/obj/scripts/py_matter_yamltests/matter_yamltests.tests.test_yaml_loader.py.stamp
[830/844] touch mac_x64_gcc/obj/scripts/py_matter_yamltests/matter_yamltests.tests.stamp
[831/844] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../src/credentials/tests --default-toolchain=//third_party/pigweed/repo/pw_toolchain/default:default --current-toolchain=//config/mbedtls/toolchain:mac_x64_gcc_mbedtls --touch mac_x64_gcc_mbedtls/gen/src/credentials/tests/TestCommissionerDUTVectors_run.pw_pystamp --capture-output --module pw_unit_test.test_runner --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files mac_x64_gcc_mbedtls/gen/src/credentials/tests/TestCommissionerDUTVectors_run_metadata_path_list.txt -- --runner ../../third_party/pigweed/repo/targets/host/run_test --test mac_x64_gcc_mbedtls/tests/TestCommissionerDUTVectors
[832/844] touch mac_x64_gcc_mbedtls/obj/src/credentials/tests/TestCommissionerDUTVectors_run.stamp
[833/844] touch mac_x64_gcc_mbedtls/obj/src/credentials/tests/tests_run.stamp
[834/844] touch mac_x64_gcc_mbedtls/obj/src/crypto_tests_run.stamp
[835/844] touch obj/check_host_gcc_mbedtls_crypto_tests.stamp
[836/844] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../src/transport/tests --default-toolchain=//third_party/pigweed/repo/pw_toolchain/default:default --current-toolchain=//build/toolchain/host:mac_x64_gcc --touch mac_x64_gcc/gen/src/transport/tests/TestSessionManager_run.pw_pystamp --capture-output --module pw_unit_test.test_runner --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files mac_x64_gcc/gen/src/transport/tests/TestSessionManager_run_metadata_path_list.txt -- --runner ../../third_party/pigweed/repo/targets/host/run_test --test mac_x64_gcc/tests/TestSessionManager
[837/844] touch mac_x64_gcc/obj/src/transport/tests/TestSessionManager_run.stamp
[838/844] touch mac_x64_gcc/obj/src/transport/tests/tests_run.stamp
ninja: build stopped: cannot make progress due to previous errors.
Error: Process completed with exit code 1.
```

andy31415, 2023-07-06:
Relevant logs seem to be:

```
[1688629967897] [16502:74420] [EM] <<< [E:35407i S:1 M:67690427] (S) Msg TX to 2:0000000011223344 [6EF6] --- Type 0004:01 (Echo:EchoRequest)
[1688629967897] [16502:74420] [IN] (S) Sending msg 67690427 on secure session with LSID: 1
[1688629967985] [16502:74420] [EM] >>> [E:35407r S:2 M:67690427] (S) Msg RX from 1:0000000012344321 [6EF6] --- Type 0004:01 (Echo:EchoRequest)
[1688629967985] [16502:74420] [EM] Handling via exchange: 35407r, Delegate: 0x7ff7ba6c8a68
[1688629967985] [16502:74420] [EM] Retransmitting MessageCounter:67690427 on exchange 35407i Send Cnt 1
[1688629967985] [16502:74420] [IN] (S) Sending msg 67690427 on secure session with LSID: 1
[1688629967985] [16502:74420] [IN] Received a duplicate message with MessageCounter:67690427 on exchange 35407r
[1688629967985] [16502:74420] [EM] >>> [E:35407r S:2 M:67690427] (S) Msg RX from 1:0000000012344321 [6EF6] --- Type 0004:01 (Echo:EchoRequest)
[1688629967985] [16502:74420] [EM] Found matching exchange: 35407r, Delegate: 0x7ff7ba6c8a68
[1688629967985] [16502:74420] [EM] Forcing tx of solitary ack for duplicate MessageCounter:67690427 on exchange 35407r
[1688629967985] [16502:74420] [EM] <<< [E:35407r S:2 M:50900557 (Ack:67690427)] (S) Msg TX to 1:0000000012344321 [6EF6] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1688629967985] [16502:74420] [IN] (S) Sending msg 50900557 on secure session with LSID: 2
[1688629967986] [16502:74420] [EM] >>> [E:35407i S:1 M:50900557 (Ack:67690427)] (S) Msg RX from 2:0000000011223344 [6EF6] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1688629967986] [16502:74420] [EM] Found matching exchange: 35407i, Delegate: 0x0
[1688629967986] [16502:74420] [EM] Rxd Ack; Removing MessageCounter:67690427 from Retrans Table on exchange 35407i
src/messaging/tests/TestReliableMessageProtocol.cpp:932: assertion failed: "loopback.mSentMessageCount == 1"
src/messaging/tests/TestReliableMessageProtocol.cpp:934: assertion failed: "rm->TestGetCountRetransTable() == 1"
'#3:','Test ReliableMessageMgr::CheckDuplicateMessage                                                                   ','FAILED'
...
'#6:','1','18'
'#7:','2','716'
INF Test 1/1: [FAIL] TestReliableMessageProtocol in 4.885 s
```

nivi-apple, 2023-07-06:
Re-ran the build job and it passed - https://github.com/project-chip/connectedhomeip/actions/runs/5472582066/jobs/9982957939. will keep an eye out for similar failure.

woody-apple, 2023-07-07:
https://github.com/project-chip/connectedhomeip/actions/runs/5485364918/jobs/9994095415

bzbarsky-apple, 2023-07-10:
The relevant test code is:
```
    // Drop the ack, and keep the exchange around to receive the duplicate message
    mockReceiver.mDropAckResponse = true;
...
    err = exchange->SendMessage(Echo::MsgType::EchoRequest, std::move(buffer));
    NL_TEST_ASSERT(inSuite, err == CHIP_NO_ERROR);
    ctx.DrainAndServiceIO();

    // Ensure the message was sent
    // The ack was dropped, and message was added to the retransmit table
    NL_TEST_ASSERT(inSuite, loopback.mSentMessageCount == 1);
```

The important part of the log here is:
```
2023-07-06T07:52:50.1624290Z [1688629967897] [16502:74420] [EM] <<< [E:35407i S:1 M:67690427] (S) Msg TX to 2:0000000011223344 [6EF6] --- Type 0004:01 (Echo:EchoRequest)
2023-07-06T07:52:50.1781930Z [1688629967985] [16502:74420] [EM] >>> [E:35407r S:2 M:67690427] (S) Msg RX from 1:0000000012344321 [6EF6] --- Type 0004:01 (Echo:EchoRequest)
...
2023-07-06T07:52:50.1983460Z [1688629967985] [16502:74420] [EM] Retransmitting MessageCounter:67690427 on exchange 35407i Send Cnt 1
2023-07-06T07:52:50.2286140Z [1688629967985] [16502:74420] [EM] >>> [E:35407r S:2 M:67690427] (S) Msg RX from 1:0000000012344321 [6EF6] --- Type 0004:01 (Echo:EchoRequest)
...
2023-07-06T07:52:50.2590240Z [1688629967985] [16502:74420] [EM] <<< [E:35407r S:2 M:50900557 (Ack:67690427)] (S) Msg TX to 1:0000000012344321 [6EF6] --- Type 0000:10 (SecureChannel:StandaloneAck)
2023-07-06T07:52:50.2792560Z [1688629967986] [16502:74420] [EM] >>> [E:35407i S:1 M:50900557 (Ack:67690427)] (S) Msg RX from 2:0000000011223344 [6EF6] --- Type 0000:10 (SecureChannel:StandaloneAck)
```

So we sent a message, then lost the timeslice for 88ms before it was processed on the receiver side; receiver responded but we ignored the ack.  By this point our MRP retry had gotten hit, so we sent the message _again_, and receiver acked.  So we got two message sends happening, no the one the test expected and asserted.

This is exactly the situation that got fixed in #27817.

Duplicate of #27810

closed 2023-07-10 by bzbarsky-apple