project-chip/connectedhomeip #27675: [ReadHandler] Merge different callbacks objects into one.

lpbeliveau-silabs, 2023-07-06:
Currentl the ReadHandler object has:
`ApplicationCallback`
`ManagementCallback`
and soon
`Observer`
That are all used for the purpose of registering observers to this class, however the ReadHandler only allows one of each.
These three objects should be merged into the `Observer` class and there should be a pool of observers instead of a single one allowed.

bzbarsky-apple, 2023-07-06:
There's a conceptual difference between ManagementCallback (which is SDK-internal) and ApplicationCallback (which is provided by the SDK consumer).  But yes, in general it might be possible to merge the two into a single type, depending.

lpbeliveau-silabs, 2023-07-11:
Also needs to make the IM engine the observer in ReadHandler Constructor, see: https://github.com/project-chip/connectedhomeip/pull/27553/files#r1260232449

lpbeliveau-silabs, 2023-07-11:
Note: SetObserver() method will be unnecessary once this issue is addressed. https://github.com/project-chip/connectedhomeip/pull/27553/files#r1260247068