project-chip/connectedhomeip #27689: [CHIPTool]: Device not found after attempting to commission the same device following unpairing

mukesh-simform, 2023-07-07:
### Reproduction steps

Run the CHIPTool sample app and follow the steps below 

1. Commission a matter-enabled device - **ESP 32-chip** with CHIPTool app
2. **Unpair device** 
3. Attempt to commission the **same device again**.

Every time I get a `Device Not Found` popup. 
And to commission that chip again I've to Flash the ESP32 chip repeatedly.

[image]

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

master

### Platform

android, esp32

### Anything else?

Very first time commissioning worked as expected but not able to commission the same device after unparing.

In order to commission the same ESP32 device I've to flash the ESP and then it works perfectly until I unpair that device.

I don't want to flash ESP32 all the time to commission that again.

nicelyjust, 2023-07-18:
Did you try to scanning BLE by other BLE Tool(like nRF ) 
Or you can try to increase the bluetooth scan time @mukesh-simform

dhrishi, 2023-07-18:
@mukesh-simform Please share the ESP32 side console logs. May be, after unpairing the device would still be on the IP network and hence you won't be able to find the device over BLE. The next commissioning should be on-network commissioning in that case. I am not sure if the CHIPTool app finds the device on all the interfaces (BLE as well as Wi-Fi)

mukesh-simform, 2023-07-18:
@dhrishi Thank you for looking into it and sharing your input. The log shows fabric was removed successfully. If the device is still in the network then what would be the way to commission that again from an implementation perspective, `pairDeviceWithAddress` will expect the IP address and that seems missing in the current scenario so how can I get that IP address If, I'm implementing that as a custom flow instead of something similar to chip tool app.

[image]
[image]

dhrishi, 2023-07-18:
>If the device is still in the network then what would be the way to commission that again from an implementation perspective,

I am not an expert of the controller implementation and do not know the exact APIs. But you can try using the host-based chip-tool by using `chip-tool pairing onnetwork` command and recommission the device

closed 2023-07-28 by mukesh-simform

mukesh-simform, 2023-08-17:
We can unpair the device using the following command 

`./chip-tool pairing unpair <node-id>`

And in chiptool android app we've equivalent method 

 `deviceController.unpairDeviceCallback(nodeId, ChipUnpairDeviceCallback())`
 
Because still device will be in network so for commissioning the  same device again we can use following command 

`./chip-tool pairing onnetwork <node-id> <setup-code>`

**Now how can we re-commission an already existing device on network using node id and setup code in Android?**

I've tried the following method but every time it throws an exception "Failed to pair the device"

` deviceController.pairDeviceWithCode(20231, "20202021", false, true, null, network)`

**Most important if device is already on network then why this method is expecting network credentials again, onnetwork command is only expecting node id and setup code**  Despite providing network credentials it still thows the following exception

```
Failed to pair the device.
Java.lang.NoSuchMethodError: no non-static method "Lchip/devicecontroller/ChipDeviceControllerException;.<init>(ILjava/lang/String;)V"
```

[image]

**There should be a detailed documentation of chip controller methods and usages, Haven't found any**