project-chip/connectedhomeip #27747: [TC-SMCO-2.6] - SDK has to be implemented for Smoke and CO Alarm Cluster

Rajashreekalmane, 2023-07-07:
### Reproduction steps

Summary

**All-clusters-app has to be enabled with below mentioned Attributes :**

- [ ]  expressed-state
- [ ]  smoke-state
- [ ]  costate
- [ ]  battery-alert
- [ ]  device-muted
- [ ]  test-in-progress
- [ ]  hardware-fault-alert
- [ ]  end-of-service-alert
- [ ]  interconnect-smoke-alarm
- [ ]  interconnect-coalarm
- [ ]  contamination-state
- [ ]  sensitivity-level
- [ ]  expiry-date

**All-clusters-app has to be enabled with below mentioned commands :**

- [ ] SelfTestRequest command

**All-clusters-app has to be enabled with below mentioned events :**

- [ ] smoke-alarm
- [ ]  coalarm
- [ ]  low-battery
- [ ]  hardware-fault
- [ ]  end-of-service
- [ ]  self-test-complete
- [ ]  alarm-muted
- [ ]  mute-ended
- [ ]  interconnect-smoke-alarm
- [ ]  interconnect-coalarm
- [ ]  all-clear

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

d55f68526864b2986404d1da9c2ccd8b64735b6e - master

### Platform

raspi

### Platform Version(s)

RPI4 8GB RAM

bzbarsky-apple, 2023-07-07:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/27746

ericzijian1994, 2023-08-23:
@Rajashreekalmane Hi, may I know what problem you still have with TC2.6?

Survensa, 2023-08-23:
The test case is fully executable; hence, I am closing this issue.

closed 2023-08-23 by Survensa