project-chip/connectedhomeip #27762: [TC-MNCONC-1.1] SDK has to be implemented for Manganese Concentration Measurement Cluster

Survensa, 2023-07-07:
### Reproduction steps

**All-clusters-app has to be enabled with below mentioned commands :**

- [ ]  generated-command-list
- [ ]  accepted-command-list
- [ ]  event-list
- [ ]  attribute-list
- [ ]  feature-map
- [ ]  cluster-revision

### Bug prevalence

Whenever I try to read all these Global attributes

### GitHub hash of the SDK that was being used

7ae5919a046a18c7e60537aff814977ee424b9cf - Master

### Platform

raspi

### Platform Version(s)

RPI - 4, 8GB RAM

mhazley, 2023-07-07:
Will be fixed by https://github.com/project-chip/connectedhomeip/issues/27662

Survensa, 2023-07-07:
Test case removed in https://github.com/CHIP-Specifications/chip-test-plans/pull/2995

closed 2023-07-07 by Survensa