project-chip/connectedhomeip #27787: [TC-TCTL-3.3] SDK has to be implemented for Temperature Control Cluster

Survensa, 2023-07-07:
### Summary

**All-clusters-app has to be enabled with below mentioned Attribute :**

- [ ] CurrentTemperatureLevel

**All-clusters-app has to be enabled with below mentioned Attributes :**

- [ ] SetTemperature

### GitHub hash of the SDK that was being used

40fb7c246671e9d37ca8bb2cfbf37f81c1509334

### Platform

raspi

### Platform Version(s)

RPI - 4, 8GB RAM

bzbarsky-apple, 2023-07-07:
Duplicate of combination of https://github.com/project-chip/connectedhomeip/issues/27786 and https://github.com/project-chip/connectedhomeip/issues/27784

jadhavrohit924, 2023-07-19:
Temperature control cluster impl added in [701fcb2](https://github.com/project-chip/connectedhomeip/commit/701fcb21f9a1bc4c7792c6e0709a3c74101fa6f5)

closed 2023-07-25 by Survensa