project-chip/connectedhomeip #27794: [TC-OPSTATE-2.2] SDK has to be implemented for Operational State Cluster

Rajashreekalmane, 2023-07-07:
### Reproduction steps

All-clusters-app has to be enabled with below mentioned Attributes :

- [ ]  PhaseList
- [ ]  CurrentPhase
- [ ]  CountdownTime
- [ ]  OperationalStateList
- [ ]  OperationalState
- [ ]  OperationalError
- [ ] Bug prevalence
- [ ] Everytime

All-clusters-app has to be enabled with below mentioned commands :

- [ ] Start
- [ ] Stop

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

d55f68526864b2986404d1da9c2ccd8b64735b6e - master

### Platform

raspi

### Platform Version(s)

RPI4 8GB RAM

bzbarsky-apple, 2023-07-07:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/27793

Rajashreekalmane, 2023-07-18:
Issue resolved in latest master TOT : b6caecf7860072bd4e55be43dc830b5ce990ddef

closed 2023-07-18 by Rajashreekalmane