project-chip/connectedhomeip #27796: [BUG] pairing code-thread failed due to Internal error

lancersky2017, 2023-07-07:
### Reproduction steps

```
$ out/host/chip-tool pairing code-thread 1003 0e080000000000010000000300001a35060004001fffe00208fafa573ad7b5b2800708fd315f3ce60abf700510cd3a3ce681ccfb77b1e5fac44b0f103b030b474c2d533230302d3466640102abee0410af223d4b4ff19bcecb5ce49e384f86c00c0402a0f7f8 MT:6FCJ142C00KA0648G00
```

```
[1688724428.742408][722006:722008] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1688724428.747282][722006:722008] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1688724428.823767][722006:722007] CHIP:DL: Indication received, conn = 0x7fbe940481b0
[1688724428.823913][722006:722008] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1688724428.824020][722006:722008] CHIP:EM: >>> [E:13420i M:90729078] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
[1688724428.824038][722006:722008] CHIP:EM: Found matching exchange: 13420i, Delegate: 0x7fbe94049d88
[1688724428.824051][722006:722008] CHIP:DMG: ICR moving to [ResponseRe]
[1688724428.824081][722006:722008] CHIP:DMG: InvokeResponseMessage =
[1688724428.824088][722006:722008] CHIP:DMG: {
[1688724428.824096][722006:722008] CHIP:DMG:    suppressResponse = false, 
[1688724428.824102][722006:722008] CHIP:DMG:    InvokeResponseIBs =
[1688724428.824116][722006:722008] CHIP:DMG:    [
[1688724428.824127][722006:722008] CHIP:DMG:            InvokeResponseIB =
[1688724428.824141][722006:722008] CHIP:DMG:            {
[1688724428.824149][722006:722008] CHIP:DMG:                    CommandDataIB =
[1688724428.824159][722006:722008] CHIP:DMG:                    {
[1688724428.824169][722006:722008] CHIP:DMG:                            CommandPathIB =
[1688724428.824196][722006:722008] CHIP:DMG:                            {
[1688724428.824210][722006:722008] CHIP:DMG:                                    EndpointId = 0x0,
[1688724428.824230][722006:722008] CHIP:DMG:                                    ClusterId = 0x31,
[1688724428.824243][722006:722008] CHIP:DMG:                                    CommandId = 0x5,
[1688724428.824253][722006:722008] CHIP:DMG:                            },
[1688724428.824269][722006:722008] CHIP:DMG: 
[1688724428.824281][722006:722008] CHIP:DMG:                            CommandFields = 
[1688724428.824295][722006:722008] CHIP:DMG:                            {
[1688724428.824310][722006:722008] CHIP:DMG:                                    0x0 = 1, 
[1688724428.824324][722006:722008] CHIP:DMG:                            },
[1688724428.824337][722006:722008] CHIP:DMG:                    },
[1688724428.824356][722006:722008] CHIP:DMG: 
[1688724428.824367][722006:722008] CHIP:DMG:            },
[1688724428.824383][722006:722008] CHIP:DMG: 
[1688724428.824393][722006:722008] CHIP:DMG:    ],
[1688724428.824416][722006:722008] CHIP:DMG: 
[1688724428.824429][722006:722008] CHIP:DMG:    InteractionModelRevision = 1
[1688724428.824438][722006:722008] CHIP:DMG: },
[1688724428.824485][722006:722008] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0005
[1688724428.824533][722006:722008] CHIP:CTL: Received NetworkConfig response, networkingStatus=1
[1688724428.824559][722006:722008] CHIP:CTL: Error on commissioning step 'ThreadNetworkSetup': '../../src/controller/CHIPDeviceController.cpp:2050: CHIP Error 0x000000AC: Internal error'
[1688724428.824567][722006:722008] CHIP:CTL: Failed to perform commissioning step 15
[1688724428.824578][722006:722008] CHIP:CTL: Going from commissioning step 'ThreadNetworkSetup' with lastErr = '../../src/controller/CHIPDeviceController.cpp:2050: CHIP Error 0x000000AC: Internal error' -> 'Cleanup'
[1688724428.824596][722006:722008] CHIP:CTL: Performing next commissioning step 'Cleanup' with completion status = '../../src/controller/CHIPDeviceController.cpp:2050: CHIP Error 0x000000AC: Internal error'
[1688724428.824606][722006:722008] CHIP:CTL: Successfully finished commissioning step 'Cleanup'
[1688724428.824620][722006:722008] CHIP:TOO: Device commissioning Failure: ../../src/controller/CHIPDeviceController.cpp:2050: CHIP Error 0x000000AC: Internal error
[1688724428.824653][722006:722008] CHIP:DMG: ICR moving to [AwaitingDe]
[1688724428.824831][722006:722006] CHIP:CTL: Shutting down the commissioner
[1688724428.824856][722006:722006] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1688724428.824865][722006:722006] CHIP:DIS: Closing all BLE connections
[1688724428.824874][722006:722006] CHIP:IN: Clearing BLE pending packets.
[1688724428.824908][722006:722006] CHIP:BLE: Auto-closing end point's BLE connection.
[1688724428.824913][722006:722006] CHIP:DL: Closing BLE GATT connection (con 0x7fbe940481b0)
[1688724428.824932][722006:722006] CHIP:IN: SecureSession[0x7fbe8c026b10]: MarkForEviction Type:1 LSID:43230
[1688724428.824941][722006:722006] CHIP:SC: SecureSession[0x7fbe8c026b10]: Moving from state 'kActive' --> 'kPendingEviction'
[1688724428.824973][722006:722006] CHIP:IN: SecureSession[0x7fbe8c026b10]: Released - Type:1 LSID:43230
[1688724428.824988][722006:722006] CHIP:CTL: Shutting down the controller
[1688724428.824996][722006:722006] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1688724428.825003][722006:722006] CHIP:FP: Forgetting fabric 0x1
[1688724428.825018][722006:722006] CHIP:TS: Pending Last Known Good Time: 2023-07-05T16:24:18
[1688724428.825126][722006:722006] CHIP:TS: Previous Last Known Good Time: 2023-07-05T16:24:18
[1688724428.825139][722006:722006] CHIP:TS: Reverted Last Known Good Time to previous value
[1688724428.825143][722006:722007] CHIP:DL: BluezDisconnect peer=E8:A8:3A:F8:42:B9
[1688724428.825159][722006:722006] CHIP:CTL: Shutting down the commissioner
[1688724428.825187][722006:722006] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1688724428.825193][722006:722006] CHIP:CTL: Shutting down the controller
[1688724428.825198][722006:722006] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1688724428.825293][722006:722006] CHIP:DMG: IM WH moving to [Uninitialized]
[1688724428.825301][722006:722006] CHIP:DMG: IM WH moving to [Uninitialized]
[1688724428.825306][722006:722006] CHIP:DMG: IM WH moving to [Uninitialized]
[1688724428.825310][722006:722006] CHIP:DMG: IM WH moving to [Uninitialized]
[1688724428.825317][722006:722006] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1688724428.825340][722006:722006] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1688724428.825361][722006:722006] CHIP:FP: Shutting down FabricTable
[1688724428.825370][722006:722006] CHIP:TS: Pending Last Known Good Time: 2023-07-05T16:24:18
[1688724428.825422][722006:722006] CHIP:TS: Previous Last Known Good Time: 2023-07-05T16:24:18
[1688724428.825429][722006:722006] CHIP:TS: Reverted Last Known Good Time to previous value
[1688724428.825576][722006:722006] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-bTrA1V)
[1688724428.826226][722006:722006] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1688724428.826247][722006:722006] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1688724428.826256][722006:722006] CHIP:DL: Inet Layer shutdown
[1688724428.826261][722006:722006] CHIP:DL: BLE shutdown
[1688724428.826743][722006:722006] CHIP:DL: System Layer shutdown
[1688724429.002985][722006:722006] CHIP:TOO: Run command failure: ../../src/controller/CHIPDeviceController.cpp:2050: CHIP Error 0x000000AC: Internal error
```

### Bug prevalence

failed all the time

### GitHub hash of the SDK that was being used

v1.0.0.2、rc2

### Platform

nrf

lancersky2017, 2023-07-07:
Need prefix `hex:` for operationalDataset
```
~# chip-tool pairing code-thread 1001 hex:0e080000000000010000000300000c35060004001fffe00208a5d7fac32c67e1230708fdd5f1bd0cb9d2910510a4c1eaa962ec9d963775149a08e0dd13030f4f70656e5468726561642d373830610102780a04100efdc647ec
66d7b625ab2a3fd60735ec0c0402a0f7f8 MT:6FCJ142C00KA0648G00
```

closed 2023-07-07 by lancersky2017

potto216, 2024-04-07:
I ran into this too which was fixed by using `hex:`. Can we improve the error message since this seems like a common issue?