project-chip/connectedhomeip #27797: android_building.md Flamingo Android Studio Java instructions

willysagefalk, 2023-07-07:
Made a fresh install of Ubuntu 22.04.2 LTS

Installed Android Studio using "software" in Ubuntu
Android Studio Flamingo | 2022.2.1 Patch 1

According to Building instruction you should set 
JAVA_HOME=/Applications/Android\ 
Studio.app/Contents/jbr/Contents/Home/

Java was not found using that path 
./scripts/build/build_examples.py --target android-arm64-chip-tool build
2023-07-07 10:20:09 INFO    Building targets: android-arm64-chip-tool
2023-07-07 10:20:09 INFO    Preparing builder 'android-arm64-chip-tool'
2023-07-07 10:20:09 INFO    Generating /home/willy/matter/connectedhomeip/out/android-arm64-chip-tool
2023-07-07 10:20:09 INFO    Setting up Android deps through Gradle
2023-07-07 10:20:09 WARNING 
2023-07-07 10:20:09 WARNING ERROR: JAVA_HOME is set to an invalid directory: /Applications/Android Studio.app/Contents/jbr/Contents/Home/
2023-07-07 10:20:09 WARNING 
2023-07-07 10:20:09 WARNING Please set the JAVA_HOME variable in your environment to match the
2023-07-07 10:20:09 WARNING location of your Java installation.

In Android Studio Flamingo I found default Java  
JAVA_HOME=/snap/android-studio/126/android-studio/jbr

When I check version it appears to be version 17.0.06 however
/snap/android-studio/126/android-studio/jbr/bin/java --version
openjdk 17.0.6 2023-01-17
OpenJDK Runtime Environment (build 17.0.6+0-17.0.6b802.4-9586694)
OpenJDK 64-Bit Server VM (build 17.0.6+0-17.0.6b802.4-9586694, mixed mode)

So unsure how to set JAVA_HOME correctly using Flamingo

Building from scripts worked without JAVA_HOME using 
sudo apt install openjdk-11-jdk-headless
sudo snap install --classic kotlin
/scripts/build/build_examples.py --target android-arm64-chip-tool build

The resulting APK runs fine on an Android Phone

luke-ingle, 2023-07-11:
I think the JAVA_HOME value is relevant to Mac OSX, not Linux.

yunhanw-google, 2023-08-09:
could you consider to use docker to complete android chip-tool
docker run -it -v ~/workspace/connectedhomeip:/connectedhomeip ghcr.io/project-chip/chip-build-android:1
git config --global --add safe.directory /connectedhomeip
git config --global --add safe.directory /connectedhomeip/third_party/pigweed/repo
source scripts/bootstrap.sh
./scripts/build/build_examples.py --target android-arm64-chip-tool build