project-chip/connectedhomeip #27805: [TC-TIMESYNC-2.7] - python script is failing at Test Step 14

Rajashreekalmane, 2023-07-07:
### Reproduction steps

Setting up the environment :
Step 1 : cloned Connectedhomeip using mentioned command : [git clone ](https://github.com/project-chip/connectedhomeip.git)https://github.com/project-chip/connectedhomeip.git
step 2 : cd connectedhomeip
step 3 : pulled latest commit
step 4: source scripts/bootstrap.sh
step 5 : ./scripts/build_python.sh --install_virtual_env  build-env
step 6 : source scripts/activate.sh

executed the below mentioned command to put DUT into a commissionable state,
./chip-all-clusters-app
Once DUT reach the commissionable state sent below mentioned command on TH.
rm -f admin_storage.json &&  python3 src/python_testing/TC_TIMESYNC_2_7.py --discriminator 3840 --passcode 20202021 --commissioning-method on-network --storage-path admin_storage.json

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

d55f68526864b2986404d1da9c2ccd8b64735b6e - master

### Platform

raspi

### Platform Version(s)

RPI4 8GB RAM

### Anything else?

[TC-TIMESYNC-2.7] test case is failing while reading LocalTime attribut  at step 14

```

[MatterTest] 07-07 18:29:02.478 ERROR Exception occurred in test_TC_TIMESYNC_2_7.
Traceback (most recent call last):
  File "/home/grl/Master/connectedhomeip/build-env/lib/python3.10/site-packages/mobly/base_test.py", line 783, in exec_one_test
    test_method()
  File "/home/grl/Master/connectedhomeip/src/python_testing/matter_testing_support.py", line 935, in async_runner
    return asyncio.run(body(*args, **kwargs))
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/home/grl/Master/connectedhomeip/src/python_testing/TC_TIMESYNC_2_7.py", line 126, in test_TC_TIMESYNC_2_7
    compare_time(received=local, offset=timedelta(seconds=7200), tolerance=timedelta(seconds=5))
  File "/home/grl/Master/connectedhomeip/src/python_testing/matter_testing_support.py", line 195, in compare_time
    asserts.assert_less_equal(delta, tolerance, "Received time is out of tolerance")
  File "/home/grl/Master/connectedhomeip/build-env/lib/python3.10/site-packages/mobly/asserts.py", line 218, in assert_less_equal
    _call_unittest_assertion(_pyunit_proxy.assertLessEqual,
  File "/home/grl/Master/connectedhomeip/build-env/lib/python3.10/site-packages/mobly/asserts.py", line 52, in _call_unittest_assertion
    raise signals.TestFailure(my_msg, extras=extras)
mobly.signals.TestFailure: Details=datetime.timedelta(seconds=3600, microseconds=2422) not less than or equal to datetime.timedelta(seconds=5) Received time is out of tolerance, Extras=None
[MatterTest] 07-07 18:29:02.481 INFO [Test] test_TC_TIMESYNC_2_7 FAIL
[MatterTest] 07-07 18:29:02.488 INFO Summary for test class TC_TIMESYNC_2_7: Error 0, Executed 1, Failed 1, Passed 0, Requested 1, Skipped 0
[MatterTest] 07-07 18:29:02.489 INFO Summary for test run MatterTest@07-07-2023_18-28-44-702:
Total time elapsed 17.786965564999264s
Artifacts are saved in "/tmp/matter_testing/logs/MatterTest/07-07-2023_18-28-44-702"
Test summary saved in "/tmp/matter_testing/logs/MatterTest/07-07-2023_18-28-44-702/test_summary.yaml"
Test results: Error 0, Executed 2, Failed 1, Passed 1, Requested 2, Skipped 0
[1688734742.490485][26162:26165] CHIP:SC: SecureSession[0x7fa264064910, LSID:46428]: State change 'kActive' --> 'kPendingEviction'
[1688734742.490517][26162:26165] CHIP:FP: Forgetting fabric 0x1
[1688734742.490544][26162:26165] CHIP:TS: Pending Last Known Good Time: 2023-07-07T17:28:18

```

**PFA logs below**
[[TC-TIMESYNC-2.7].txt](https://github.com/project-chip/connectedhomeip/files/11982241/TC-TIMESYNC-2.7.txt)

cecille, 2023-07-07:
I think this is an implementation issue, not a test issue. IIRC, expiring the time zone and offsets wasn't yet implemented. Test step 14 also only happens if the time zone list size is > 1. I thought the plan was to reduce the TZ and DST list max sizes to 1 to get the tests going. @fessehaeve will have more info.

cecille, 2023-07-07:
Actually, nope, I was wrong. Please disregard last comment. There exists a fix for this already, it is in https://github.com/project-chip/connectedhomeip/pull/27545/files

closed 2023-07-17 by mergify[bot]