project-chip/connectedhomeip #27819: Add Darwin API to implement session resumption storage

bzbarsky-apple, 2023-07-07:
We don't want to require Matter.framework consumers to use MTRStorage to handle session resumption.

The plan is to introduce a new delegate interface that can perform the following operations:

1. Map a (controller, peer node id) pair to session resumption info: resumption id, shared secret, peer CATS.  This is needed to do resumption as a CASE initiator.
2. Map a "resumption id" to session resumption info: shared secret, peer CATs, peer node id, controller.  This is needed to do resumption as a CASE responder.
3. Store resumption info for a session: input is controller, shared secret, peer CATs, peer node id, resumption id.
4. Clear all resumption info for a given controller.

Given operation 2, this needs to be provided on the MTRDeviceControllerFactory.

Implementations that don't want to handle operation 2 can then just use the passed-in controller (available for all the other operations) to defer to per-controller objects as desired.

bzbarsky-apple, 2023-09-01:
As part of https://github.com/project-chip/connectedhomeip/pull/28533 we moved session resumption storage to be handled by the new storage API.  This does not allow a completely custom session resumption implementation, but no longer uses MTRStorage and breaks up the resumption storage into per-controller storage buckets.

If custom resumption implementation is needed, it could be done on a per-controller basis, with operation 2 supported in the framework by looping over all running controllers and asking them for the relevant info.

woody-apple, 2023-10-27:
@bzbarsky-apple to evaluate if this is needed anymore.

bzbarsky-apple, 2023-10-27:
Going to close this for now.  MTRDeviceControllerStorageDelegate works for the use cases I am aware of. Once we have someone asking for something fully custom we can figure out what the API should be.

closed 2023-10-27 by bzbarsky-apple