project-chip/connectedhomeip #27821: [Build] linux tests failing

woody-apple, 2023-07-07:
### Build issue(s)

https://github.com/project-chip/connectedhomeip/actions/runs/5488103763/jobs/10000479690

bzbarsky-apple, 2023-07-08:
```

ERROR   16:21:18.536 - TEST ERR  : WARNING: ThreadSanitizer: thread leak (pid=21073)
ERROR   16:21:18.536 - TEST ERR  :   Thread T2 (tid=21083, finished) created by main thread at:
ERROR   16:21:18.536 - TEST ERR  :     #0 pthread_create ../staging/llvm_build/tools/clang/stage2-bins/runtimes/runtimes-x86_64-unknown-linux-gnu-bins/compiler-rt/lib/tsan/rtl/tsan_interceptors_posix.cpp:1019:3 (chip-tool+0x2e494b) (BuildId: a35653ef98fcc100)
ERROR   16:21:18.537 - TEST ERR  :     #1 chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_StartEventLoopTask() /__w/connectedhomeip/connectedhomeip/out/linux-x64-chip-tool-no-ble-tsan-clang/../../examples/chip-tool/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_POSIX.ipp:294:11 (chip-tool+0x171cb19) (BuildId: a35653ef98fcc100)
ERROR   16:21:18.537 - TEST ERR  :     #2 chip::DeviceLayer::PlatformManager::StartEventLoopTask() /__w/connectedhomeip/connectedhomeip/out/linux-x64-chip-tool-no-ble-tsan-clang/../../examples/chip-tool/third_party/connectedhomeip/src/include/platform/PlatformManager.h:416:44 (chip-tool+0x18e0535) (BuildId: a35653ef98fcc100)
```

This is actually a fairly common random failure on Linux, from what I hae seen....

andy31415, 2023-08-24:
Is this still common? Our CI builds seem more green recently.

andy31415, 2023-10-10:
I have not seen this in a long time. Marking fixed.

closed 2023-10-10 by andy31415