project-chip/connectedhomeip #27823: Attribute persistence provider API change issues

bzbarsky-apple, 2023-07-07:
@hicklin Please see https://github.com/project-chip/connectedhomeip/pull/27611#pullrequestreview-1520069631

@tcarmelveilleux @tobiasgraf 

If we're going to make an API change here, we should think pretty hard about what the API should look like instead of taking EmberAfAttributeMetadata....  The type+size may not be the right thing.

Plus the various other issues.  And the fact that this breaks API compat for SDK consumers.

hicklin, 2023-07-07:
What we can do is overload the `ReadValue` function with the same signature as before and make that function call the new `ReadValue`. That should solve to issue of SDK consumers calling a function that currently does not exist.

bzbarsky-apple, 2023-07-08:
> That should solve to issue of SDK consumers calling a function that currently does not exist.

No, you misunderstand.  The SDK consumers _implement_ this function.  This is the API you implement, as an SDK consumer, to handle attributes being written.  You can leave it as the "thing that uses the KVS", or you can implement your own persistence provider.  And in the latter case you will now no longer compile, because we broke backwards compat.

hicklin, 2023-07-28:
Fixed [here](https://github.com/project-chip/connectedhomeip/pull/28302).

bzbarsky-apple, 2023-07-28:
Unfortunately, not fixed @hicklin.  Specifically, https://github.com/project-chip/connectedhomeip/pull/27611#discussion_r1256564193 is not addressed yet.

PR #28302 (2023-07-28): Reverted the AttributePersistenceProvider and added a SafeAttributePersistenceProvider
`Fixes #27823`

There are two main concerns with the current implementation of the AttributePersistenceProvider.

1. The `AttributePersistenceProvider::ReadValue` method signature has changed. It is expected that some products in the field will have implemented their own method and this change will brake those implementations. This is not a big issue since the change from the manufacturer would be minimal but it would be nice to avoid it.
2. The current implementation assumes that all data in the storage is  [... cut]