project-chip/connectedhomeip #27824: [CERT-BLOCKER] Additional Logging not appearing in TH side logs even when using --trace_decode 1 on 3c8b8c SHA

Garik098, 2023-07-08:
### Reproduction steps

1 -> Build accessory side apps (All Cluster app, TV app, etc) on SHA 3c8b8cf9680e0715d50a9866b125effd7530dbd5 on Raspi platform
2 -> While running the app include " --trace_decode 1 " to enable additional logging (Few test cases need this additional logging to check few values)
3 -> Observe that the TH side logs doesn't show additional logging even when we running the app with --trace_decode 1

Few test cases have usage of --trace_decode 1 as prerequisite and few test steps in the test cases use the additional logging to verify the outcome. So having additional logging while using this flag is essential for evaluating the test cases. 

### Bug prevalence

All the time

### GitHub hash of the SDK that was being used

3c8b8cf9680e0715d50a9866b125effd7530dbd5

### Platform

darwin, raspi

liamgonyea, 2023-07-08:
Sample of all-cluster-app logs pairing on 2 different SHAs

SHA mentioned above with  --trace_decode not working: 3c8b8cf
[missing_trace_decode_3c8b8cf968.txt](https://github.com/project-chip/connectedhomeip/files/11987037/missing_trace_decode_3c8b8cf968.txt)

Old SHA when --trace_decode was working: df4ad952 
[working_trace_decode_df4ad952.txt](https://github.com/project-chip/connectedhomeip/files/11987038/working_trace_decode_df4ad952.txt)

bzbarsky-apple, 2023-07-08:
@andy31415 This looks like it was caused by https://github.com/project-chip/connectedhomeip/pull/27135; could you take a look please?

bzbarsky-apple, 2023-07-08:
Reverting https://github.com/project-chip/connectedhomeip/commit/5ed31da5cc5086fcfe0c3271c046ecbc87c30970 seems to make things work again.

andy31415, 2023-07-08:
Will take a look. We seemed to started depending on logging encrypted buffer content and that does not look right, however if we need to add more data to post-decryption side that is probably ok.

andy31415, 2023-07-08:
@Garik098 please provide a list of things that certification relies upon and is only available in trace decode. we have to either fix trace decode or make that data available in some form of stable/documented format for "make sure it keeps being available"

andy31415, 2023-07-10:
Looking to reproduce this on linux, assuming realistically all apps and chip-tool should be the same. Will test with `linux-x64-light-no-ble-with-ui` and `linux-x64-chip-tool`

andy31415, 2023-07-10:
I believe `trace-decode` should generally apply to chip-tool (because we can control that) and not the app (since when we test black box devices, we would not have access to logs). @Garik098 could you confirm? Your instructions describe logging in the all-clusters app.

andy31415, 2023-07-10:
ran:

```
./out/linux-x64-chip-tool/chip-tool pairing code 1 34970112332 --trace_decode 1 2>&1 | tee ~/tmp/chip-tool-pairing-log.txt
```

Inside that file I see some decoded commands, however I do not see attributes (not 0x3e 1 reads).

andy31415, 2023-07-10:
On linux I could also do: `./out/linux-x64-chip-tool/chip-tool pairing code 1 34970112332 --trace-to json:$HOME/tmp/chip-tool-pairing-trace.json` and that shows me that 0x3e == 62 == OperationalCredentials.
I see:

OperationalCredentials::CertificateChainRequest
OperationalCredentials::CertificateChainResponse 
OperationalCredentials::AttestationRequest
OperationalCredentials::AttestationResponse
OperationalCredentials::CSRRequest
OperationalCredentials::CSRResponse
OperationalCredentials::AddTrustedRootCertificate (2x)
OperationalCredentials::AddNOC
OperationalCredentials::NOCResponse

All of the above seem to be commands, never attribute reads. I will see if I can manually trigger and attribute read . Attribute 1 there seems to be "0x0001 == Fabrics".

andy31415, 2023-07-10:
Ok, so the instructions did say `./darwin-framework-tool operationalcredentials read fabrics 1 0` ... will try the same.

andy31415, 2023-07-10:
At least logging to json shows the data, so it is available

[image]

Will try trace decoding to see what that does

andy31415, 2023-07-10:
I see trace_decode as well (however this is chip-tool, not the app):

[image]

I think I will probably need to test with the app (assuming we are trying to validate chip-tool implementations)

andy31415, 2023-07-10:
This is also the response, but not the reply. I assume chip-tool has to send a request first which I do not see.

andy31415, 2023-07-10:
I think it may be that tracedecoder is stateful: https://github.com/project-chip/connectedhomeip/blob/master/examples/common/tracing/TraceDecoder.cpp#L100

So tracing "encrypted" data seems to actually also trace unencrypted data that was saved during a previous call.

bzbarsky-apple, 2023-07-10:
> and not the app (since when we test black box devices, we would not have access to logs)

In this case the DUT is the client, so that's the black box, and the app is part of the TH.

> I think it may be that tracedecoder is stateful

Yes, that's what I was telling you on Slack multiple times now....

andy31415, 2023-07-10:
> > I think it may be that tracedecoder is stateful
> 
> Yes, that's what I was telling you on Slack multiple times now....

Ok, will follow through the code paths to see what gets things to output what we want.

andy31415, 2023-07-10:
We have a revert open as well as a "fix to show the traces". Current worry about the fix is that MRP retries would stop recording data (as trace decoder does not remember anything now) so need to determine if we need MRP retries to keep human readable output somehow (and if so, how that is used).

Garik098, 2023-07-10:
> I believe `trace-decode` should generally apply to chip-tool (because we can control that) and not the app (since when we test black box devices, we would not have access to logs). @Garik098 could you confirm? Your instructions describe logging in the all-clusters app.

For Accessory side testing, chip tool logs are what we look into. For the Controller side tests, we relay on the app side logs since Controllers won't share their logs for review during events or certification. Since the access to the Controller side logs were not available, few testcases mention to use --trace_decode to get additional logging to validate the step.

Garik098, 2023-07-10:
> @Garik098 please provide a list of things that certification relies upon and is only available in trace decode. we have to either fix trace decode or make that data available in some form of stable/documented format for "make sure it keeps being available"

Multiple testcases use trace_decode on the Controller side testing since we don't give the Controller side logs and the only way to validate is to look at the app logs. It is spread across core, app cluster testcases and I think it would be better to have trace_decode enabled rather than working on the individual ones.

andy31415, 2023-07-10:
What I am looking at is what kind of information needs to be available. I see in this case we are looking at payload data. Are there other fields that have explicit test usage? 

Wondering which fields of the trace decode are used and how, since I seem to have unknowingly broken the tests, would like to be able to double check next time.

PR #27860 (2023-07-11): Make tracedecoder not stateful and ensure both sent and received messages are logged
#27276 mistakenly assumed trace decoding was not stateful (i.e. logging a JSON would just output that json). Instead tracing was preserving values across calls, so that only one of the 2 logs was outputing data (and tracing of encrypted payloads would preserve data from unencrypted ones).

Changes:
  - make trace decoder not be stateful
  - add inbound/outbound flags for tracing unencrypted payloads
  - add peer address for unencrypted payload sending (since this data was picked up from sending itself)

Fixes  [... cut]
closed 2023-07-11 by mergify[bot]