project-chip/connectedhomeip #27833: [CHIP-TOOL] Error occurs while compiling CHIP-TOOL

Survensa, 2023-07-10:
### Reproduction steps

1. gn gen out/debug --args='chip_mdns="platform" chip_inet_config_enable_ipv4=false'
2. ninja -C out/debug

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

488aabfe70bb3fcbf22ca77767899e55f6c0a030

### Platform

raspi

### Platform Version(s)

RPI-4, 8GB-RAM variant

### Anything else?

### Error log :

```
ninja: Entering directory `out/debug'
[294/1165] c++ obj/BUILD_DIR/gen/src/controller/d...tlv/meta/cluster-tlv-metadata.clusters_meta.cpp.o
FAILED: obj/BUILD_DIR/gen/src/controller/data_model/tlv/meta/cluster-tlv-metadata.clusters_meta.cpp.o
aarch64-linux-gnu-g++ -MMD -MF obj/BUILD_DIR/gen/src/controller/data_model/tlv/meta/cluster-tlv-metadata.clusters_meta.cpp.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -Igen/src/controller/data_model -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -c gen/src/controller/data_model/tlv/meta/clusters_meta.cpp -o obj/BUILD_DIR/gen/src/controller/data_model/tlv/meta/cluster-tlv-metadata.clusters_meta.cpp.o
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp: In function ‘void __static_initialization_and_destruction_0(int, int)’:
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:861:23: error: statement has no effect [-Werror=unused-value]
  861 | const Entry<ItemInfo> _BasicInformation_ShutDown[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:2197:23: error: statement has no effect [-Werror=unused-value]
 2197 | const Entry<ItemInfo> _TimeSynchronization_DSTTableEmpty[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:2209:23: error: statement has no effect [-Werror=unused-value]
 2209 | const Entry<ItemInfo> _TimeSynchronization_TimeFailure[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:2212:23: error: statement has no effect [-Werror=unused-value]
 2212 | const Entry<ItemInfo> _TimeSynchronization_MissingTrustedTimeSource[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:2306:23: error: statement has no effect [-Werror=unused-value]
 2306 | const Entry<ItemInfo> _BridgedDeviceBasicInformation_ShutDown[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:2309:23: error: statement has no effect [-Werror=unused-value]
 2309 | const Entry<ItemInfo> _BridgedDeviceBasicInformation_Leave[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:3240:23: error: statement has no effect [-Werror=unused-value]
 3240 | const Entry<ItemInfo> _SmokeCoAlarm_SmokeAlarm[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:3243:23: error: statement has no effect [-Werror=unused-value]
 3243 | const Entry<ItemInfo> _SmokeCoAlarm_COAlarm[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:3246:23: error: statement has no effect [-Werror=unused-value]
 3246 | const Entry<ItemInfo> _SmokeCoAlarm_LowBattery[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:3249:23: error: statement has no effect [-Werror=unused-value]
 3249 | const Entry<ItemInfo> _SmokeCoAlarm_HardwareFault[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:3252:23: error: statement has no effect [-Werror=unused-value]
 3252 | const Entry<ItemInfo> _SmokeCoAlarm_EndOfService[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:3255:23: error: statement has no effect [-Werror=unused-value]
 3255 | const Entry<ItemInfo> _SmokeCoAlarm_SelfTestComplete[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:3258:23: error: statement has no effect [-Werror=unused-value]
 3258 | const Entry<ItemInfo> _SmokeCoAlarm_AlarmMuted[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~
[... 44 lines cut ...]
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:4727:23: error: statement has no effect [-Werror=unused-value]
 4727 | const Entry<ItemInfo> _PumpConfigurationAndControl_ElectronicNonFatalFailure[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:4730:23: error: statement has no effect [-Werror=unused-value]
 4730 | const Entry<ItemInfo> _PumpConfigurationAndControl_ElectronicFatalFailure[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:4733:23: error: statement has no effect [-Werror=unused-value]
 4733 | const Entry<ItemInfo> _PumpConfigurationAndControl_GeneralFault[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:4736:23: error: statement has no effect [-Werror=unused-value]
 4736 | const Entry<ItemInfo> _PumpConfigurationAndControl_Leakage[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:4739:23: error: statement has no effect [-Werror=unused-value]
 4739 | const Entry<ItemInfo> _PumpConfigurationAndControl_AirDetection[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gen/src/controller/data_model/tlv/meta/clusters_meta.cpp:4742:23: error: statement has no effect [-Werror=unused-value]
 4742 | const Entry<ItemInfo> _PumpConfigurationAndControl_TurbineOperation[] = {
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
[299/1165] c++ obj/src/controller/tests/data_model/libDataModelTests.TestRead.cpp.o
ninja: build stopped: subcommand failed.
```

tianfeng-yang, 2023-07-10:
I also got the same error

vediyappan-hcl, 2023-07-11:
I also got the same error. I have included the changes added in "https://github.com/project-chip/connectedhomeip/pull/27839/files", it still shows one active error:

```
vediyappan@vediyappan-ThinkPad-P15-Gen-1:~/work/matter/new/connectedhomeip$ gn gen out/debug
Done. Made 6858 targets from 342 files in 191ms
vediyappan@vediyappan-ThinkPad-P15-Gen-1:~/work/matter/new/connectedhomeip$ ninja -C out/debug
ninja: Entering directory `out/debug'
[39/1515] c++ obj/src/lib/format/tests/libFormatTests.TestDecoding.cpp.o
**FAILED: obj/src/lib/format/tests/libFormatTests.TestDecoding.cpp.o**
g++ -MMD -MF obj/src/lib/format/tests/libFormatTests.TestDecoding.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -Igen/src/controller/data_model -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -Igen/src/lib/format -I../../third_party/nlunit-test/repo/src -c ../../src/lib/format/tests/TestDecoding.cpp -o obj/src/lib/format/tests/libFormatTests.TestDecoding.cpp.o
../../src/lib/format/tests/TestDecoding.cpp: In function ‘void __static_initialization_and_destruction_0(int, int)’:
../../src/lib/format/tes**ts/TestDecoding.cpp:37:23: error: statement has no effect [-Werror=unused-value]
   37 | const Entry<ItemInfo> _empty_item[0]                 = {};
      |                       ^~~~~~~~~~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
[52/1515] c++ obj/src/lib/support/tests/libSupportTests.TestPersistedCounter.cpp.o
ninja: build stopped: subcommand failed.**
vediyappan@vediyappan-ThinkPad-P15-Gen-1:~/work/matter/new/connectedhomeip$
```

closed 2023-07-11 by woody-apple