project-chip/connectedhomeip #27856: [Feature] Door Bell

newcomertv, 2023-07-10:
### Feature description

As far as I can tell Matter already supports, buttons, notifications and speakers.
The logical step in a modern smart home would be to be able to configure a button at the front of the entrance door or wherever I might want to trigger a ring to ring on a list of predefined speakers.

As far as I can tell there is no explicit support for doorbells or their upgrade(s) -> intercom or even better video intercom

The current hacky way to support such a use case would be to either have the button be a notifiable event (like an occupancy sensor) or have the button start playing a certain music on a given speaker starting a timer and stopping once the timer is elapsed.

Both options seem suboptimal.

Even better would be the option to use a givens smart speaker and microphone for audio intercomms.

If the TV is on, a chosen cameras video feed could pop into the screen as well.

### Platform

all

bzbarsky-apple, 2023-07-10:
@newcomertv This sounds like a feature request for the spec, right?  If so, it needs to be brought to the CSA as part of the standards process....

newcomertv, 2023-07-11:
My bad. I recently discovered the feature request tag and milestone trackers for future revisions. I thought it might be a policy change for more transparency and collaboration with the community. I must have missed them before.

I'm closing this issue now, honouring the CSA's standards process and the contribution guidelines...

closed 2023-07-11 by newcomertv