project-chip/connectedhomeip #27880: [Platform] ota-requester app quits abruptly after doing back to back BDX transfer (when no apply argument set)

woody-apple, 2023-07-11:
### Reproduction steps

[Matter SWU] ota-requester app quits abruptly after doing back to back BDX transfer (when no apply argument set)

Steps To Reproduce:

1. Run ota-requester app on the terminal but don’t run the Apply argument (-a ), see below command used:
./chip-ota-requestor-app --discriminator 100 --KVS ota_req10 -p 5 --secured-device-port 5560

2. Pair ota-requester in Home app
3. Select Update button and see that the BDX transfer is done completely and since there is no apply argument -a on the command it does not apply the update so then it triggers the BDX transfer again. 
4. From the logs it did seem like BDX transfer occurred fully 2 times and on the 3rd attempt BDX transfer abruptly quit mid way due to below error. 

[1682369652802] [19631:284411] [EM] Rxd Ack; Removing MessageCounter:73244494 from Retrans Table on exchange 8418i
[1682369652802] [19631:284411] [SPT] VerifyOrDie failure at ../../examples/ota-requestor-app/linux/third_party/connectedhomeip/src/app/clusters/ota-requestor/ota-requestor-server.cpp:190: EMBER_ZCL_STATUS_SUCCESS == status
Abort trap: 6

### Platform

darwin

### Platform Version(s)

1.2

### Type

Manually tested with SDK

nivi-apple, 2023-07-11:
I tested using the ota-requestor-app on the mac without auto apply update and do not see a crash. will set up the home app to try and see if i can reproduce the crash.

nivi-apple, 2023-07-25:
Was able to reproduce on a setup for ios. Will debug further

PR #28309 (2023-07-27): Reset the downloadedBytes and totalFileBytes in the OTAImageProgress …
…params to 0 when we prepare for download for BDX

Fixes: #27880
closed 2023-07-27 by mergify[bot]