project-chip/connectedhomeip #27903: iOS commissioning in our

LucaFagan, 2023-07-11:
### Feature description

We are trying to implement an iOS app that commissions devices in our ecosystem.

NOC should be provided by our server. 

So questions are:

1) what are the steps to do that ?
2) do I need to implement MTROperationalCertificateIssuer protocol ?
3) do I have to use createCertificateSigningRequest to create a CSR that is used by the app to get the NOC from our server ?
4) what else do we need to implement ?

Sorry for asking a lot of questions but it's not easy for use to understand the flow.

### Platform

darwin

### Platform Version(s)

1.0

bzbarsky-apple, 2023-09-01:
> what are the steps to do that ?

@LucaFagan Provide a MTROperationalCertificateIssuer, have that issue the relevant certificates.  Otherwise, just do commissioning as normal...  So implement `MTRDeviceControllerDelegate`, use `setupCommissioningSessionWithPayload` on the controller, when you get the `controller:commissioningSessionEstablishmentDone:` notification use `commissionNodeWithID` on the controller to commission.

> do I need to implement MTROperationalCertificateIssuer protocol ?

If you want to issue the NOC yourself, instead of having the framework create one using a root key it's given, then yes.

> do I have to use createCertificateSigningRequest to create a CSR that is used by the app to get the NOC from our server ?

The CSR is created by the device being commissioned.  It will be handed to the `MTROperationalCertificateIssuer` during the commissioning process, as part of the `MTROperationalCSRInfo`.

No worries about the questions.  This all needs more documentation.  That's being worked on, but doesn't exist yet.

Let me know whether the above addresses your issues, please.

LucaFagan, 2023-09-05:
Hello,
thanks for the reply.

We have created a class for instatiating the controller, root certificate and iOS device NOC are provided by our server.

```
import Foundation

class MatterService
{
    static let shared = MatterService()
    let vendorId = 0xFFF1
    
    var matterController: MTRDeviceController?
    var rootCertificateDer: Data?
    var controllerNocDer: Data?
    
    init()
    {
        // _ = self.start()
        _ = self.startWithCertificates()
        NSLog("MatterService->init")
    }
    
    func start() -> Bool
    {
        if (self.matterController == nil)
        {
            let storage = MatterStorage.shared
            let factory = MTRDeviceControllerFactory.sharedInstance()
            let factoryParams = MTRDeviceControllerFactoryParams(storage: storage)
            
            do
            {
                try factory.start(factoryParams)
            }
            catch
            {
                return false
            }
            
            let keys = FabricKeys()
            let params = MTRDeviceControllerStartupParams(signing: keys, fabricId: 1, ipk: keys.ipk)
            params.vendorID = NSNumber(value: self.vendorId)
            
            self.matterController = try? factory.createController(onExistingFabric: params)
            
            if (self.matterController == nil)
            {
                self.matterController = try? factory.createController(onNewFabric: params)
            }
        }
        
        return (self.matterController != nil)
    }
    
    func startWithCertificates() -> Bool
    {
        if (self.matterController == nil)
        {
            // LF: get root certificate
            guard let rootCertificatePem: String = try? KeychainRegistryService.shared.get(key: Constants.RegistryKeychain.rootCertificate) else
            {
                return false
            }
            
            self.rootCertificateDer = MatterCertificateHelper.convertPemToDerData(permCertificate: rootCertificatePem)

            // LF: get controller noc
            guard let controllerNocPem: String = try? KeychainRegistryService.shared.get(key: Constants.RegistryKeychain.controllerNoc) else
[... 261 lines cut ...]
                    self.bindTermController(deviceNid: self.deviceNid, zoneEndpoint: zoneEndpoint)
                    
                case .failure(let error):
                    OSUIKit.showMessageBox(viewController: self,
                                           title: Localizer.Generic.error.localized,
                                           message: "Non è stato possibile aggiornare il dispositivo sul cloud:\n\n" + error.localizedDescription,
                                           cancelButtonTitle: nil,
                                           destructiveButtonTitle: Localizer.Generic.ok.localized,
                                           otherButtonTitles: nil,
                                           tapBlock: nil)
                    
                    print(error.localizedDescription)
            }
        }
    }
```

All MTRDeviceControllerDelegate callbacks are called and the device is sucessfully commissioned.

shouldSkipAttestationCertificateValidation is called but issueOperationalCertificate is not called.

What is wrong in my code ?

Regards
Luca

robinmo, 2023-09-19:
Hi @LucaFagan 
Our team case is more or less the same:
1. we get rcac, icac, app node id and iot node id from our server.
2. our app act as a commissioner to add the iot into our ecosystem(fabric), on the commission stage, the app issue noc to iot.

> your case noc from server, this one maybe a little complex, need to get publickey from csr then request from server for noc signing, the worry is that the between request and response maybe sometime a long time(if network is terrible), it cannot be guaranteed the commission window is still open.

woody-apple, 2024-05-06:
Closing pending further feedback based on the solution above.

closed 2024-05-06 by woody-apple