project-chip/connectedhomeip #27907: [BUG] darwin-framework-tool test fails in Smoke CO Alarm

ericzijian1994, 2023-07-12:
### Reproduction steps

Since the Smoke CO Alarm cluster is added to the provisional list in availability.yaml, no relevant code is generated, but test_cluster.zapt still generates code that results in undeclared identifiers.

[logs.txt](https://github.com/project-chip/connectedhomeip/files/12026349/logs.txt)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

7fc3ccba35d4df7d2d3ba8603ff72d218af0b1a7

### Platform

darwin

bzbarsky-apple, 2023-07-14:
The test should be added to the "disable" section in examples/darwin-framework-tool/templates/tests/ciTests.json with comments similar to the other tests that are disabled there.

ericzijian1994, 2023-07-15:
> The test should be added to the "disable" section in examples/darwin-framework-tool/templates/tests/ciTests.json with comments similar to the other tests that are disabled there.

I've already disabled them here in #27123, but it doesn't work.

I see `{{#if (isSupported (asUpperCamelCase name preserveAcronyms=true))}}` statements in other zapt files, but the test_cluster.zapt file that is causing the error does not have this code.

bzbarsky-apple, 2023-07-15:
> I've already disabled them here in https://github.com/project-chip/connectedhomeip/pull/27123, but it doesn't work.

That's because you didn't disable all of them, no?  I only see 2_2 through 2_6 listed under the disabled bit, butwhat about 2_1 and 1_1?

> but the test_cluster.zapt file that is causing the error does not have this code.

That's correct, because it can't tell what would need to be supported for which YAML file.

I am working on having darwin-framework-tool be able to do all the bits, including "provisional" ones, but for now please just add them to that "disabled" list.

ericzijian1994, 2023-07-17:
I see, thanks.

closed 2023-07-17 by ericzijian1994