project-chip/connectedhomeip #27908: [CHIP-TOOL] Error occurs while compiling CHIP-TOOL

Survensa, 2023-07-12:
### Reproduction steps

1. gn gen out/debug --args='chip_mdns="platform" chip_inet_config_enable_ipv4=false'
2. ninja -C out/debug

### Bug prevalence

After the PR : https://github.com/project-chip/connectedhomeip/pull/27839

### GitHub hash of the SDK that was being used

9bf0fe6c63979701b4344d2fce1a54ca8bdbf0a3

### Platform

raspi

### Platform Version(s)

RPI-4, 8GB-RAM variant

### Anything else?

```
ninja: Entering directory `out/debug'
[619/2140] c++ obj/src/lib/format/tests/libFormatTests.TestDecoding.cpp.o
FAILED: obj/src/lib/format/tests/libFormatTests.TestDecoding.cpp.o
aarch64-linux-gnu-g++ -MMD -MF obj/src/lib/format/tests/libFormatTests.TestDecoding.cpp.o.d -Wconversion -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DCHIP_HAVE_CONFIG_H=1 -Igen/src/controller/data_model -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/repo/include -Igen/src/lib/format -I../../third_party/nlunit-test/repo/src -c ../../src/lib/format/tests/TestDecoding.cpp -o obj/src/lib/format/tests/libFormatTests.TestDecoding.cpp.o
../../src/lib/format/tests/TestDecoding.cpp: In function ‘void __static_initialization_and_destruction_0(int, int)’:
../../src/lib/format/tests/TestDecoding.cpp:37:23: error: statement has no effect [-Werror=unused-value]
   37 | const Entry<ItemInfo> _empty_item[0]                 = {};
      |                       ^~~~~~~~~~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
cc1plus: all warnings being treated as errors
[624/2140] c++ obj/src/lib/shell/shell_core.Engine.cpp.o
ninja: build stopped: subcommand failed.
```

andy31415, 2023-07-14:
tried on vscode image, it compiled fine. Code looks like:

```
const Entry<ItemInfo> _empty_item[0]                 = {};
const std::array<const Node<ItemInfo>, 1> empty_meta = { { { 0, _empty_item } } };
```

So it is used. Will need to reproduce to see what makes this compiler happy.

andy31415, 2023-07-14:
@Survensa what ubuntu version did you run on the raspberry pi?

andy31415, 2023-07-14:
As a temporary workaround, you could set `chip_build_tests=false` in the gn gen arguments. This should also make the compile faster while still compiling chip-tool.

Still trying to reproduce, however RPi compilations are massively slow.

andy31415, 2023-07-14:
Seems to work for me:

```
andrei@pidevel:~/devel/connectedhomeip$ ninja -C out/debug/ TestDecoding
ninja: Entering directory `out/debug/'
[18/18] ld tests/TestDecoding
andrei@pidevel:~/devel/connectedhomeip$ uname -a
Linux pidevel 5.11.0-1027-raspi #30-Ubuntu SMP PREEMPT Thu Jan 13 12:52:48 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
andrei@pidevel:~/devel/connectedhomeip$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=21.04
DISTRIB_CODENAME=hirsute
DISTRIB_DESCRIPTION="Ubuntu 21.04"
```

andy31415, 2023-07-14:
@Survensa - need environment info: what O/S version are you using to compile and maybe `aarch64-linux-gnu-g++ --version`

PR #27973 (2023-07-14): Do not allow zero size array in test
Apparently this is not allowed by C/C++ standard and some compilers are just lenient.

I believe this fixes #27908 (however I could not actually reproduce that bug ... so just assuming it fixes it)
closed 2023-07-14 by mergify[bot]