project-chip/connectedhomeip #27913: [BUG] chip-tool 'Can not decode data' thread stable-data-version 0xff

q-thla, 2023-07-12:
### Reproduction steps

"chip-tool threadnetworkdiagnostics read stable-data-version" fails to decode when the value returned is 255 (the UINT8_MAX).

This issue is seen in certification test TC_DGTHREAD_2_2 sometimes.

This attribute is mentioned by the matter specification as "0x000c StableDataVersion uint8", which seems to match with what is in the codebase:

src/app/zap-templates/zcl/data-model/chip/thread-network-diagnostics-cluster.xml:    <attribute side="server" code="0x0C" define="STABLE_DATA_VERSION" type="**INT8U**" min="0x00" max="0xFF" writable="false"  isNullable="true" optional="false">StableDataVersion</attribute>
third_party/zap/repo/zcl-builtin/matter/thread-network-diagnostics-cluster.xml:    <attribute side="server" code="0x0C" define="STABLE_DATA_VERSION" type="**INT8U**" min="0x00" max="0xFF" writable="false" optional="false">StableDataVersion</attribute>

### Bug prevalence

seen twice during testing session

### GitHub hash of the SDK that was being used

raspberry pi test harness v2.8.1

### Platform

raspi

### Anything else?

[image]

bzbarsky-apple, 2023-07-12:
In the spec this is a nullable uint8.  That means that the allowed range is 0-254, and 255 is not a valid value.

If 255 is actually supposed to be a valid value for this field, the spec needs to change or something...

andy31415, 2023-07-12:
Probably a similar question on other fields in `11.13.6. Attributes`. For example DataVersion seems very related and is also a nullable uint8 (so range 0..254).

jwhui, 2023-07-12:
All values in range [0, 255] are valid values for the Thread Network Data versions.

bzbarsky-apple, 2023-07-12:
I filed https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/7301 for DataVersion and StableDataVersion, but this issue may affect other Thread Network Diagnostics attributes as well....