project-chip/connectedhomeip #27920: [ReadHandler] Unify Now timestamps

lpbeliveau-silabs, 2023-07-12:


bzbarsky-apple, 2023-07-12:
The reason this matters: we can get radically different (hundreds of milliseconds apart) values for two consecutive now() calls on Mac/Linux, and the code very much does not expect that to happen as it's written now.

lpbeliveau-silabs, 2023-07-14:
See here:
https://github.com/project-chip/connectedhomeip/pull/27553/files#r1260284902
https://github.com/project-chip/connectedhomeip/pull/27553/files#r1260287402

bzbarsky-apple, 2023-07-14:
@lpbeliveau-silabs For future reference, links to comments in the "files" view go dead when the code changes...  You probably want:

* https://github.com/project-chip/connectedhomeip/pull/27553#discussion_r1260284902
* https://github.com/project-chip/connectedhomeip/pull/27553#discussion_r1260287402

to link to the comments in the main discussion view... (note that the comment ids are the same).

lpbeliveau-silabs, 2023-08-14:
Solved by https://github.com/project-chip/connectedhomeip/pull/28536

lpbeliveau-silabs, 2023-08-14:
https://github.com/project-chip/connectedhomeip/pull/28536