project-chip/connectedhomeip #27932: [Info Wants to Know] Matter supports on some router

AllsVoid, 2023-07-13:
Hi guys~
Recently I am following the steps given by the [matter website](https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app) to build a matter application for connecting IoT devices, here is my device:
- x86_64 OpenWRT router
- nRF52840 dk & nRF52840 Dongle

At first, I established a thread network connection with nRF52840 dk by installing openthread on OpenWRT with nRF52840 dongle 
> like OpenWRT+nRF52840 dongle(ot-rcp.hex) <---> nRF52840 dk+PC

Later, I want to try if I can connect or control the device through the Matter Controller ([chip-tool Android app](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/nrfconnect_android_commissioning.md)) without plugging in another device on OpenWRT.

Therefore, I want to know:
- Is it possible to install the Matter application on OpenWRT, connect to OpenWRT through Matter Controller, and then control the function of the device (for example, control the LED lights of OpenWRT through matter controller)
- If there is no hardware supported by Matter on OpenWRT, can the router's own hardware be used instead?

Thanks in advance to anyone who answered my questions~

closed 2023-08-23 by AllsVoid

testuser7, 2023-09-29:
There is an implementation of Matter Controller Server in python: https://github.com/home-assistant-libs/python-matter-server

QingtianHoorii, 2024-01-11:
hi Klee，did you compile any matter app successfully on openwrt?

AllsVoid, 2024-01-11:
> hi Klee，did you compile any matter app successfully on openwrt?

Klee if fine~
Yes , i successfully compile a matter app on OpenWRT.
However, I give up the dongle pairing method, use the SDK `examples/lighting-app/linux`, and the compiled program can run on OpenWRT

Remember to use the right OpenWRT Toolchain. 
> For example arch x86_64 -> OpenWRT_xxx_x86_64_xxx

QingtianHoorii, 2024-01-12:
Glad to hear you finished compiling，I thought you might be interested in experimenting with this  [repo](https://github.com/project-chip/matter-openwrt),After a series of trials, I  got a chip-tool which could use it pairing with a Thread dongle on OpenWrt。