project-chip/connectedhomeip #27937: [Build] TI applications fail to build (intelhex)

jjm202, 2023-07-13:
### Build issue(s)

```

 2045  git clone https://github.com/project-chip/connectedhomeip.git
 2046  cd connectedhomeip/
 2047  git submodule update --init
 2048  source ./scripts/bootstrap.sh
 2049  cd examples/pump-controller-app/cc13x2x7_26x2x7/
 2050  gn gen out/debug --args="ti_sysconfig_root=\"$HOME/ti/sysconfig_1.15.0\""
 2051  ninja -C out/debug

ninja: Entering directory `out/debug'
[1/9] ACTION //:pump_controller_app.bin(//third_party/connectedhomeip/build/toolchain/arm_gcc:arm_gcc)
FAILED: chip-LP_CC2652R7-pump-controller-example.bin
python ../../third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//third_party/con
nnectedhomeip/build/toolchain/arm_gcc:arm_gcc --capture-output --python-virtualenv-config python/gen/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/pigweed_buil
adata_path_list.txt -- ../../third_party/connectedhomeip/third_party/ti_simplelink_sdk/repo_cc13xx_cc26xx/tools/common/oad/oad_image_tool.py --verbose ccs . 7 -hex1 chip-LP_CC2652
../third_party/connectedhomeip/third_party/ti_simplelink_sdk/repo_cc13xx_cc26xx/tools/common/oad/private.pem
Traceback (most recent call last):
  File "../../third_party/connectedhomeip/third_party/ti_simplelink_sdk/repo_cc13xx_cc26xx/tools/common/oad/oad_image_tool.py", line 22, in <module>
    import intelhex
ModuleNotFoundError: No module named 'intelhex'
ninja: build stopped: subcommand failed.
```

### Platform

cc13x2_cc26x2

### Anything else?

```
jmarkel@ILA1LT7WDN593:~/connectedhomeip/examples/pump-controller-app/cc13x2x7_26x2x7$ pip install intelhex
Requirement already satisfied: intelhex in /home/jmarkel/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages (2.3.0)
```

jjm202, 2023-07-13:
Interestingly, checking out a prior commit and building will work, as will switch back to the original/master (if the out/debug directory remains intact):

```
$ git checkout 66b2d01b0f47c2952495e88a95c2d6913b83e0e8
jmarkel@ILA1LT7WDN593:~/temp/connectedhomeip/examples/pump-controller-app/cc13x2x7_26x2x7$ ninja -C out/debug
ninja: Entering directory `out/debug'
[0/1] Regenerating ninja files
[402/402] stamp obj/default.stamp

jmarkel@ILA1LT7WDN593:~/temp/connectedhomeip/examples/pump-controller-app/cc13x2x7_26x2x7$ **git switch -**
Previous HEAD position was 66b2d01b0f [OIS] Platform improvements (#27291)
Switched to branch 'master'
**Your branch is up to date with 'origin/master'.**
jmarkel@ILA1LT7WDN593:~/temp/connectedhomeip/examples/pump-controller-app/cc13x2x7_26x2x7$ ninja -C out/debug
ninja: Entering directory `out/debug'
[0/1] Regenerating ninja files
[400/400] stamp obj/pump_controller_app.ota.stamp
```

but removing and regenerating the out/debug folder 
jmarkel@ILA1LT7WDN593:~/temp/connectedhomeip/examples/pump-controller-app/cc13x2x7_26x2x7$ rm -rf out/
jmarkel@ILA1LT7WDN593:~/temp/connectedhomeip/examples/pump-controller-app/cc13x2x7_26x2x7$ gn gen out/debug --args="ti_sysconfig_root=\"$HOME/ti/sysconfig_1.15.0\""
Done. Made 272 targets from 226 files in 95ms
jmarkel@ILA1LT7WDN593:~/temp/connectedhomeip/examples/pump-controller-app/cc13x2x7_26x2x7$ ninja -C out/debug
ninja: Entering directory `out/debug'
[1076/1084] ACTION //:pump_controller_app.bin(//third_party/connectedhomeip/build/toolchain/arm_gcc:arm_gcc)
FAILED: chip-LP_CC2652R7-pump-controller-example.bin
python ../../third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../.. --default-toolchain=//
onnectedhomeip/build/toolchain/arm_gcc:arm_gcc --current-toolchain=//third_party/connectedhomeip/build/toolchain/arm_gcc:arm_gcc --capture-output --python-virtualen
on/gen/third_party/connectedhomeip/third_party/pigweed/repo/pw_env_setup/pigweed_build_venv/venv_metadata.json --python-dep-list-files gen/pump_controller_app.bin_m
list.txt -- ../../third_party/connectedhomeip/third_party/ti_simplelink_sdk/repo_cc13xx_cc26xx/tools/common/oad/oad_image_tool.py --verbose ccs . 7 -hex1 chip-LP_CC
ontroller-example.hex -o chip-LP_CC2652R7-pump-controller-example -k ../../third_party/connectedhomeip/third_party/ti_simplelink_sdk/repo_cc13xx_cc26xx/tools/common
pem
Traceback (most recent call last):
  File "../../third_party/connectedhomeip/third_party/ti_simplelink_sdk/repo_cc13xx_cc26xx/tools/common/oad/oad_image_tool.py", line 22, in <module>
    import intelhex
ModuleNotFoundError: No module named 'intelhex'
ninja: build stopped: subcommand failed.

jjm202, 2023-07-14:
I worked through this with @ReneJosefsen yesterday. He suggested the following workaround, which was successful:

As a workaround, it seems like you can add:
`intelhex`
to connectedhomeip/scripts/setup/requirements.ti.txt and then add
`pw_build_PIP_REQUIREMENTS += [ "${chip_root}/scripts/setup/requirements.ti.txt" ]`
to `connectedhomeip/examples/lock-app/cc13x2x7_26x2x7/.gn`, just below` import("//args.gni")`

Also, please query TI for when the CC26x2x7 builds will be enabled again in the workflows.

ReneJosefsen, 2023-07-14:
We should find a way to get this added to the TI platform in a way that is get applied for all applicable examples. The above is mainly to get unblocked. Given this is related to OTA it could also be worth to only require these modules if OTA is enabled.

ReneJosefsen, 2023-10-27:
Given the mentioned PR have merged, I would assume this issue is ok to close.

Alami-Amine, 2024-08-09:
it looks like the issue was resolved
additionally, we are now supporting cc13x4_26x4 instead of cc13x2_26x2, I assume we can close this

closed 2024-08-09 by Alami-Amine