project-chip/connectedhomeip #27947: [Documentation] Enum field name in SDK doesn't match the name in the spec

Vyrastas, 2023-07-13:
### Documentation issues

For the OnOff cluster, the SDK uses TogglePreviousOnOff in the OnOffStartUpOnOff enum: https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/onoff-cluster.xml#L24

However in the spec it uses Toggle: https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/OnOff.adoc#ref_StartUpOnOffEnum

Not a breaking issue but worth keeping in sync.  The SDK probably should be updated to match.

@andy31415 @bzbarsky-apple

bzbarsky-apple, 2023-07-13:
See https://github.com/project-chip/connectedhomeip/issues/27092 -- the spec for this cluster could not be aligned with the XML at all until recently.  At this point we need to update the XML to match the spec, but no one has had a chance to do it yet.

Vyrastas, 2023-07-14:
> See #27092 -- the spec for this cluster could not be aligned with the XML at all until recently. At this point we need to update the XML to match the spec, but no one has had a chance to do it yet.

Ah great, thanks for letting me know.

closed 2023-09-22 by mergify[bot]