project-chip/connectedhomeip #27953: [Darwin-tool BUG] DelayedActionTime field not showing in OTA-Requestor App

liamgonyea, 2023-07-14:
### Reproduction steps

When performing an OTA transfer using Darwin-framework-tool, the `DelayedActionTime` parameter is not being set for the OTA-Requestor to wait between attempts to query the image. Instead, the default 120 seconds is being used instead of the DelayedActionTime=180 seconds value set.

### Reproduce:

Build Darwin-framework-tool on Mac using newer SHA c7d9a11dc9369bceb08beb99b0b8c0ac3d1901e7

1. `scripts/examples/gn_build_example.sh examples/darwin-framework-tool examples/darwin-framework-tool/out/debug`

Build OTA-Requestor app on RasPi using SHA 5a21d17fd2bb0a48f4a356937f4741f6c9dc7975

2. `scripts/examples/gn_build_example.sh examples/ota-requestor-app/linux out/ota-requestor-app chip_config_network_layer_ble=false chip_code_pre_generated_directory=\"/home/ubuntu/sve2/connectedhomeip/zzz_pregenerated\"`

Launch OTA-Requestor app on RasPi:

3. `./chip-ota-requestor-app --discriminator 3841 --autoApplyImage --trace_decode 1` 

Run the following commands on Mac:

4. ` ./darwin-framework-tool interactive start --commissioner-nodeId 0x1234`
5. `otasoftwareupdateapp candidate-file-path <Path_to_JSON_Config_File.json>`
6. `pairing code 4321 MT:-24J0IRV01KA0648G00`
7. `otasoftwareupdaterequestor write default-otaproviders '[{"fabricIndex": 1, "providerNodeID": "0x0000000000001234", "endpoint": 0}]' 4321 0`
8. `otasoftwareupdateapp set-reply-params --action 0 --status 1 --consent 0 --consentNeeded 0 --delayedActionTime 180`
9. `otasoftwareupdaterequestor announce-otaprovider 0x0000000000001234  65521 0 0 4321 0`

### Observe Issue in OTA-Requestor app logs

> [1689292923.476450][1173:1173] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0029 Command=0x0000_0001
[1689292923.476530][1173:1173] CHIP:SWU: QueryImageResponse:
[1689292923.476573][1173:1173] CHIP:SWU:   status: 1
[1689292923.476640][1173:1173] CHIP:SWU:   imageURI: /Users/liamgonyea/Desktop/Liam_Current_Copy/SU_BDX_TestCases/hello_OTA_v2
[1689292923.476685][1173:1173] CHIP:SWU:   softwareVersion: 2
[1689292923.476723][1173:1173] CHIP:SWU:   softwareVersionString: 2.0
[1689292923.476763][1173:1173] CHIP:SWU:   updateToken: 32
[1689292923.476804][1173:1173] **CHIP:SWU: Scheduling a retry; delay: 120**

#### In Contrast...
Repeating the same steps above except Step 1, build Darwin-framework-tool on older SHA 3c8b8cf968, yields the correct logs on the OTA-Requestor App:

> [1689292692.291065][1144:1144] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0029 Command=0x0000_0001
[1689292692.291168][1144:1144] CHIP:SWU: QueryImageResponse:
[1689292692.291213][1144:1144] CHIP:SWU:   status: 1
[1689292692.291272][1144:1144] **CHIP:SWU:   delayedActionTime: 180 seconds**
[1689292692.291313][1144:1144] CHIP:SWU:   imageURI: /Users/liamgonyea/Desktop/Liam_Current_Copy/SU_BDX_TestCases/hello_OTA_v2
[1689292692.291354][1144:1144] CHIP:SWU:   softwareVersion: 2
[1689292692.291393][1144:1144] CHIP:SWU:   softwareVersionString: 2.0
[1689292692.291450][1144:1144] CHIP:SWU:   updateToken: 32
[1689292692.291493][1144:1144] **CHIP:SWU: Scheduling a retry; delay: 180**

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

c7d9a11dc9369bceb08beb99b0b8c0ac3d1901e7

### Platform

darwin

### Anything else?

Darwin-framework-tool on newer SHA c7d9a11dc9369bceb08beb99b0b8c0ac3d1901e7 run against RasPi OTA-Requestor App SHA 5a21d17fd2bb0a48f4a356937f4741f6c9dc7975
[DelayedActionTime_Failure_c7d9a11.zip](https://github.com/project-chip/connectedhomeip/files/12045576/DelayedActionTime_Failure_c7d9a11.zip)

Darwin-framework-tool on older SHA 3c8b8cf968 run against RasPi OTA-Requestor App SHA 5a21d17fd2bb0a48f4a356937f4741f6c9dc7975
[DelayedActionTime_Success_3c8b8cf968.zip](https://github.com/project-chip/connectedhomeip/files/12045577/DelayedActionTime_Success_3c8b8cf968.zip)

bzbarsky-apple, 2023-07-14:
> `otasoftwareupdateapp candidate-file-path <Path_to_JSON_Config_File.json>`

@liamgonyea what goes in that file?

bzbarsky-apple, 2023-07-14:
Looks like https://github.com/project-chip/connectedhomeip/blob/118448d030a78a574cfbb9df1f53e0bb248e3fc8/examples/darwin-framework-tool/README.md has some docs.

bzbarsky-apple, 2023-07-14:
But those docs are not sufficient to actually set up a working JSON file, unfortunately.

bzbarsky-apple, 2023-07-18:
Per offline discussion, this never worked; @liamgonyea was using a build with some local changes initially...

krypton36, 2023-09-11:
Hello @bzbarsky-apple I believe this [PR](https://github.com/project-chip/connectedhomeip/pull/22277) fixes this issue.

PR #29169 (2023-10-19): Fix issue #27953 where OTA Requestor is not receiving delayed action …
Summary:
During query image of OTA, darwin-framework-tool (DFT) was not sending the delay action time. This PR updates the code to send the parameter in the response.

Fixes #27953

- Reproduced issue
- Tested with OTA Request as described in issue and validated issue was resolved
closed 2023-10-19 by woody-apple