project-chip/connectedhomeip #27964: [1.0]  chip-tool discover commissionables does not discover any devices

topraveenpaul, 2023-07-14:
### Reproduction steps

1. The matter device is being advertised over BLE and is discoverable by Alexa and Google home apps however, the chip-tool doesnt detect any devices.
2. Rebooted an unpaired matter device (E.g; lighting-app flashed on esp32 board) and ensure it is advertising
3. Executed the chip-tool command as below. The command just doesnt detect the device.
4.
5. $ **~/connectedhomeip/out/chip-tool discover commissionables**
[1689344873.500867][37860:37860] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1689344873.506354][37860:37860] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1689344873.506617][37860:37860] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1689344873.506722][37860:37860] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1689344873.507091][37860:37860] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-vgVaVJ)
[1689344873.507663][37860:37860] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1689344873.507716][37860:37860] CHIP:DL: NVS set: chip-counters/reboot-count = 209 (0xD1)
[1689344873.508456][37860:37860] CHIP:DL: Failed to get Ethernet interface
[1689344873.508951][37860:37860] CHIP:DL: Failed to reset Ethernet statistic counts
[1689344873.509292][37860:37860] CHIP:DL: Got WiFi interface: wlan0
[1689344873.511159][37860:37860] CHIP:DL: Found the primary WiFi interface:wlan0
[1689344873.511279][37860:37860] CHIP:IN: UDP::Init bind&listen port=0
[1689344873.511390][37860:37860] CHIP:IN: UDP::Init bound to port=58857
[1689344873.511414][37860:37860] CHIP:IN: UDP::Init bind&listen port=0
[1689344873.511503][37860:37860] CHIP:IN: UDP::Init bound to port=37717
[1689344873.511524][37860:37860] CHIP:IN: BLEBase::Init - setting/overriding transport
[1689344873.511541][37860:37860] CHIP:IN: TransportMgr initialized
[1689344873.511570][37860:37860] CHIP:FP: Initializing FabricTable from persistent storage
[1689344873.511705][37860:37860] CHIP:TS: Last Known Good Time: 2023-03-02T09:31:36
[1689344873.513190][37860:37860] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x7D19139E3B72C573, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1689344873.515957][37860:37860] CHIP:ZCL: Using ZAP configuration...
[1689344873.517029][37860:37860] CHIP:CTL: System State Initialized...
[1689344873.517088][37860:37860] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1689344873.517145][37860:37860] CHIP:CTL: Setting attestation nonce to random value
[1689344873.517192][37860:37860] CHIP:CTL: Setting CSR nonce to random value
[1689344873.517251][37860:37860] CHIP:IN: UDP::Init bind&listen port=5550
[1689344873.517342][37860:37860] CHIP:IN: UDP::Init bound to port=5550
[1689344873.517364][37860:37860] CHIP:IN: UDP::Init bind&listen port=5550
[1689344873.517435][37860:37860] CHIP:IN: UDP::Init bound to port=5550
[1689344873.517453][37860:37860] CHIP:IN: TransportMgr initialized
[1689344873.517712][37860:37865] CHIP:DL: CHIP task running
[1689344873.518138][37860:37865] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1689344873.518182][37860:37865] CHIP:CTL: Setting attestation nonce to random value
[1689344873.518343][37860:37865] CHIP:CTL: Setting CSR nonce to random value
[1689344873.519190][37860:37865] CHIP:CTL: Generating NOC
[1689344873.520039][37860:37865] CHIP:FP: Validating NOC chain
[1689344873.522029][37860:37865] CHIP:FP: NOC chain validation successful
[1689344873.522219][37860:37865] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1689344873.522256][37860:37865] CHIP:TS: Last Known Good Time: 2023-03-02T09:31:36
[1689344873.522282][37860:37865] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1689344873.522304][37860:37865] CHIP:TS: Retaining current Last Known Good Time
[1689344873.525187][37860:37865] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1689344873.531037][37860:37865] CHIP:TS: Committing Last Known Good Time to storage: 2023-03-02T09:31:36
[1689344873.532760][37860:37865] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x0000000000000000
[1689344873.532820][37860:37865] CHIP:IN: UDP::Init bind&listen port=5550
[1689344873.532952][37860:37865] CHIP:IN: UDP::Init bound to port=5550
[1689344873.532980][37860:37865] CHIP:IN: UDP::Init bind&listen port=5550
[1689344873.533077][37860:37865] CHIP:IN: UDP::Init bound to port=5550
[1689344873.533103][37860:37865] CHIP:IN: TransportMgr initialized
[1689344903.518224][37860:37860] CHIP:CTL: Shutting down the commissioner
[1689344903.518311][37860:37860] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1689344903.518569][37860:37860] CHIP:CTL: Shutting down the controller
[1689344903.518631][37860:37860] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1689344903.518682][37860:37860] CHIP:FP: Forgetting fabric 0x1
[1689344903.518750][37860:37860] CHIP:TS: Pending Last Known Good Time: 2023-03-02T09:31:36
[1689344903.519034][37860:37860] CHIP:TS: Previous Last Known Good Time: 2023-03-02T09:31:36
[1689344903.519097][37860:37860] CHIP:TS: Reverted Last Known Good Time to previous value
[1689344903.519181][37860:37860] CHIP:CTL: Shutting down the commissioner
[1689344903.519233][37860:37860] CHIP:CTL: Stopping commissioning discovery over DNS-SD
[1689344903.519449][37860:37860] CHIP:CTL: Shutting down the controller
[1689344903.519501][37860:37860] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1689344903.519887][37860:37860] CHIP:DMG: IM WH moving to [Uninitialized]
[1689344903.519944][37860:37860] CHIP:DMG: IM WH moving to [Uninitialized]
[1689344903.519990][37860:37860] CHIP:DMG: IM WH moving to [Uninitialized]
[1689344903.520035][37860:37860] CHIP:DMG: IM WH moving to [Uninitialized]
[1689344903.520085][37860:37860] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1689344903.520293][37860:37860] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1689344903.520838][37860:37860] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-fHWU7Y)
[1689344903.521828][37860:37860] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1689344903.521919][37860:37860] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1689344903.521958][37860:37860] CHIP:DL: Inet Layer shutdown
[1689344903.521990][37860:37860] CHIP:DL: BLE shutdown
[1689344903.522024][37860:37860] CHIP:DL: System Layer shutdown
[1689344903.522166][37860:37860] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:513: CHIP Error 0x00000032: Timeout

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

v1.0.0.2-2132-gdf986c96bf

### Platform

other

### Platform Version(s)

ubuntu 20.04 LTS

### Type

Platform Issue

desperadoduck, 2023-07-22:
same here. hcitool lescan on the same device finds the matter device immediatly, so it should not be an issue with bluetooth scanning.

topraveenpaul, 2023-08-02:
Strangely, not just the chip-tool, even the python controller is unable to discover these devices. At the same time, the alex/ google home in my Android is able to discover these devices automatically.

Has it ever worked? I searched all over the internet but none seems to have reported this problem. I am in need of device discovery.. Any help????

topraveenpaul, 2023-11-07:
@bzbarsky-apple

Can someone shed some light on this issue?
 It's simple and straightforward command that does not produce the expected result as mentioned in the chip sdk guide.

What are the conditions for the command "chip-tool discover commissionables" to detect the commisionable devices?

I am in a situation where I need to detect the commisionable devices that  advertises over BLE and go ahead with the pairing of that devices.

I have a nano leaf bulb that is in pairing mode.
I run the chip-tool in my raspberyr pi 4 where i have configured the chip environment. My expectation from the chip-tool is to detect the nano leaf bulb as a commissionable device so that i can go ahead and continute with pairing steps.

Any help is really appreciated.

bzbarsky-apple, 2023-11-13:
I don't know what light there is to shed.  The issue is still open.  Anyone is welcome to fix it.  No one is stepping up to do it.

dasarathg68, 2024-03-09:
Hey so I also faced this issue with ``` ./chip-tool discover commissionables ```command.
Could you check which port your device's services are running on?
The default one is 5540 but when we execute```./chip-tool discover commissionables``` the UDP port listens to 5550 (based on your output it looks like you're listening to 5550 as well).
So no commissionable devices are discovered on the network.

dasarathg68, 2024-03-09:
And sometimes ```rm -fr /tmp/chip_*``` does the trick too!

BenSumetvarapa, 2024-10-24:
Hey also facing the issue where no commissionables are returned on scan. I can see the device advertising itself through Bluetooth using bluetoothctl scan but chip-tool can't find it.
If i connect the device to google home first and connect to wifi then run discover again, I'm now able to see the device.
Is this a limitation on chip-tool discover not being able to discover over blutooth?
Has anyone else been able to discover devices over bluetooth?

bzbarsky-apple, 2024-11-01:
>  Is this a limitation on chip-tool discover not being able to discover over blutooth?

Yes, on Linux.  On Mac, it should work.

BenSumetvarapa, 2024-11-01:
> > Is this a limitation on chip-tool discover not being able to discover over blutooth?
> 
> Yes, on Linux. On Mac, it should work.

Thanks @bzbarsky-apple 
I was running chip-tool on mac and could not discover any commissionables.
I'm assuming if I can see the device over bluetooth scan I should be able to discover it with chip-tool too?

bzbarsky-apple, 2024-11-05:
@vivien-apple knows more about this, but I believe chip-tool should be able to discover over bluetooth on Mac, yes.  Depending on your compile options, possibly, and on whether you have the profile described at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#profile-installation installed.