project-chip/connectedhomeip #27982: [Groups] Failure response for group commands.

manjunath-grl, 2023-07-17:
### Reproduction steps

Summary: According to test-plan test ACE-1.6 step 7: `AddGroup` command with groupID set to 0x0104 expected outcome is `no response` but got `failure: src/transport/SessionManager.cpp:194: CHIP Error 0x000000AC: Internal error` when executed.

Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:ACE_Groups_ChipError_Response

Reproduction step:
Terminal 1:
  1. Build All-cluster-app 
       `scripts/examples/gn_build_example.sh      examples/all-clusters-app/linux/      examples/all-clusters-app/linux/out/all-clusters-app   chip_inet_config_enable_ipv4=false`
  2. Run ./chip-all-clusters-app
Terminal 2:
  1. Build Chip-tool :
       `gn gen out/debug --args='chip_mdns="platform" chip_inet_config_enable_ipv4=false' 
        ninja -C out/debug`
  2. Commission DUT and TH :
       ` ./out/debug/chip-tool pairing code 0x12344321 MT:-24J0AFN00KA0648G00`
  3. Run ACE-1.6 test:
       `./scripts/tests/yaml/chiptool.py tests Test_TC_ACE_1_6`

### Bug prevalence

Everytime when executed

### GitHub hash of the SDK that was being used

b6caecf7860072bd4e55be43dc830b5ce990ddef

### Platform

other

### Anything else?

Test-plan:
[image]
Execution logs:
[image]
[ACE_Fail_logs.zip](https://github.com/project-chip/connectedhomeip/files/12067019/ACE_Fail_logs.zip)

manjunath-grl, 2023-07-17:
Closing the ticket. Duplicate of https://github.com/project-chip/connectedhomeip/issues/27967

manjunath-grl, 2023-08-07:
Still facing issue attached logs for reference:
Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:ACE_Groups_ChipError_Response

Execution logs:
[ACE_Fail_logs.txt](https://github.com/project-chip/connectedhomeip/files/12273321/ACE_Fail_logs.txt)

Screenshots:
[image]
[image]

sumaky, 2023-09-20:
Step 7 is trying to add new group of ID 261( 0X105) using an existing group ID (0x104) that does not have privilege
Test plan does not check for any outcome and moves on with the next step.
The objective is to check no groupid of 0x105 should be added and this is done via ViewGroup using unicast method at later point of time.
The script needs to ignore as there is no expected outcome and needs to continue with the next steps.

Please let me know if there is any issue in continuing with the next steps?

cecille, 2023-09-21:
Does this error occur with the current manual steps? Or does this happen when you try to implement the yaml?

manjunath-grl, 2023-09-22:
From @sumaky comment we can understand the there wouldn't be response for the group command and script needs to ignore and proceed with next steps. But while implementing in YAML, even though we don't mention any expected response by default it expects to be  'SUCCESS' if nothing is specified.

cecille, 2023-09-22:
Yeah, the problem here is that the group command is failing in the controller because the TH doesn't have an entry for that key. It's a tooling problem. Have a patch out to suma for testing.

closed 2023-12-20 by mergify[bot]