project-chip/connectedhomeip #27988: [CERT-TEST-FAILURE] TC-TIMESYNC-2_1 - Test failed

Rajashreekalmane, 2023-07-17:
### Feature Area

Area:TimeSync

### Test Case

TC-TIMESYNC-2_1

### Reproduction steps

**Setting up the environment :**
Step 1 : cloned Connectedhomeip using mentioned command : [git clone ](https://github.com/project-chip/connectedhomeip.git)https://github.com/project-chip/connectedhomeip.git
step 2 : cd connectedhomeip
step 3 : pulled latest commit
step 4: source scripts/bootstrap.sh
step 5 : ./scripts/build_python.sh --install_virtual_env  build-env
step 6 : source scripts/activate.sh

executed the below mentioned command to put DUT into a commissionable state,
`./chip-all-clusters-app`
Once DUT reach the commissionable state sent below mentioned command on TH.
 `python3 src/python_testing/TC_TIMESYNC_2_1.py --discriminator 3840 --passcode 20202021 --commissioning-method on-network --storage-path admin_storage.json`

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

b6caecf7860072bd4e55be43dc830b5ce990ddef - MASTER

### Platform

raspi

### Anything else?

Test case is failing while reading  Granularity attribute at step 2
```
[MatterTest] 07-17 18:14:14.655 INFO ==========> TC_TIMESYNC_2_1 <==========
[MatterTest] 07-17 18:14:14.655 INFO [Test] test_TC_TIMESYNC_2_1
[MatterTest] 07-17 18:14:14.656 INFO ***** Test Step 1 : Commissioning, already done
[MatterTest] 07-17 18:14:14.656 INFO ***** Test Step 2 : Read Granularity attribute
[MatterTest] 07-17 18:14:14.656 ERROR Exception occurred in test_TC_TIMESYNC_2_1.
Traceback (most recent call last):
  File "/home/grl/Master_cntr/connectedhomeip/build-env/lib/python3.10/site-packages/mobly/base_test.py", line 783, in exec_one_test
    test_method()
  File "/home/grl/Master_cntr/connectedhomeip/src/python_testing/matter_testing_support.py", line 935, in async_runner
    return asyncio.run(body(*args, **kwargs))
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/home/grl/Master_cntr/connectedhomeip/src/python_testing/TC_TIMESYNC_2_1.py", line 46, in test_TC_TIMESYNC_2_1
    asserts.assert_true(False, "Granularity is a mandatory attribute and must be present in the PICS file")
  File "/home/grl/Master_cntr/connectedhomeip/build-env/lib/python3.10/site-packages/mobly/asserts.py", line 353, in assert_true
    fail(msg, extras)
  File "/home/grl/Master_cntr/connectedhomeip/build-env/lib/python3.10/site-packages/mobly/asserts.py", line 475, in fail
    raise signals.TestFailure(msg, extras)
mobly.signals.TestFailure: Details=Granularity is a mandatory attribute and must be present in the PICS file, Extras=None
[MatterTest] 07-17 18:14:14.657 INFO [Test] test_TC_TIMESYNC_2_1 FAIL
[MatterTest] 07-17 18:14:14.658 INFO Summary for test class TC_TIMESYNC_2_1: Error 0, Executed 1, Failed 1, Passed 0, Requested 1, Skipped 0
[MatterTest] 07-17 18:14:14.658 INFO Summary for test run MatterTest@07-17-2023_18-14-14-399:
Total time elapsed 0.25867381100033526s
Artifacts are saved in "/tmp/matter_testing/logs/MatterTest/07-17-2023_18-14-14-399"
```

**PFA log below**
[TC_TIMESYNC_2_1.txt](https://github.com/project-chip/connectedhomeip/files/12069481/TC_TIMESYNC_2_1.txt)
[TC-TIMESYNC_2.1 reference app log.txt](https://github.com/project-chip/connectedhomeip/files/12069483/TC-TIMESYNC_2.1.reference.app.log.txt)

cecille, 2023-07-17:
The problem here is that the PICS file hasn't been updated
mobly.signals.TestFailure: Details=Granularity is a mandatory attribute and must be present in the PICS file, Extras=None

cecille, 2023-07-17:
To supply a PICS file on the command line for python, use --PICS flag

cjandhyala, 2023-07-26:
closing per @Rajashreekalmane comments