project-chip/connectedhomeip #28005: [BUG] Error on TC-DRLK-2.8 execution (CONSTRAINT_ERROR)

ccruzagralopes, 2023-07-17:
### Reproduction steps

1. Run `chip-all-clusters-app`
2. Use `chip-tool` to execute the test steps for `TC-DRLK-2.8`:

Commission DUT:
```
./chip-tool pairing code 1 MT:-24J0AFN00KA0648G00
```
Step 1:
```
./chip-tool doorlock read number-of-total-users-supported 1 1
```
Step 2:
```
./chip-tool doorlock set-user 0 1 xxx 6452 1 0 0 1 1  --timedInteractionTimeoutMs 1000
```
Step 3:
```
./chip-tool doorlock get-user 1 1 1
```
Step 4:
```
./chip-tool doorlock set-user 0 2 xxx 6451 1 10 3 1 1 --timedInteractionTimeoutMs 1000
```
Step 5:
```
./chip-tool doorlock set-user 0 1 xxx 8965 1 0 0 1 1 --timedInteractionTimeoutMs 1000
```
Step 6a:
```
./chip-tool doorlock set-user 0 2 null null null null null 1 1 --timedInteractionTimeoutMs 1000
```
Step 6b:
```
./chip-tool doorlock get-user 2 1 1
```
Step 7:
```
./chip-tool doorlock set-user 2 2 null null null null null 1 1 --timedInteractionTimeoutMs 1000
```
Step 8:
```
./chip-tool doorlock get-user 2 1 1
```
Step 9:
```
./chip-tool doorlock clear-user 1 1 1  --timedInteractionTimeoutMs 1000
```
Step 10:
```
./chip-tool doorlock get-user 1 1 1  --timedInteractionTimeoutMs 1000
```
Decommission DUT:
```
./chip-tool pairing unpair 1
```

## Error

On step 4, the expected response is:
```
[1657607954.930591][7988:7993] CHIP:DMG: {
[1657607954.930618][7988:7993] CHIP:DMG:        suppressResponse = false,
[1657607954.930651][7988:7993] CHIP:DMG:        InvokeResponseIBs =
[1657607954.930688][7988:7993] CHIP:DMG:        [
[1657607954.930716][7988:7993] CHIP:DMG:                InvokeResponseIB =
[1657607954.930755][7988:7993] CHIP:DMG:                {
[1657607954.930787][7988:7993] CHIP:DMG:                        CommandStatusIB =
[1657607954.930825][7988:7993] CHIP:DMG:                        {
[1657607954.930862][7988:7993] CHIP:DMG:                                CommandPathIB =
[1657607954.930900][7988:7993] CHIP:DMG:                                {
[1657607954.930942][7988:7993] CHIP:DMG:                                        EndpointId = 0x1,
[1657607954.930984][7988:7993] CHIP:DMG:                                        ClusterId = 0x101,
[1657607954.931027][7988:7993] CHIP:DMG:                                        CommandId = 0x1a,
[1657607954.931065][7988:7993] CHIP:DMG:                                },
[1657607954.931107][7988:7993] CHIP:DMG:
[1657607954.931142][7988:7993] CHIP:DMG:                                StatusIB =
[1657607954.931180][7988:7993] CHIP:DMG:                                {
[1657607954.931218][7988:7993] CHIP:DMG:                                        status = 0x85 (INVALID_COMMAND),
[1657607954.931259][7988:7993] CHIP:DMG:                                },
[1657607954.931296][7988:7993] CHIP:DMG:
[1657607954.931333][7988:7993] CHIP:DMG:                        },
[1657607954.931371][7988:7993] CHIP:DMG:
[1657607954.931402][7988:7993] CHIP:DMG:                },
[1657607954.931439][7988:7993] CHIP:DMG:
[1657607954.931466][7988:7993] CHIP:DMG:        ],
[1657607954.931500][7988:7993] CHIP:DMG:
[1657607954.931527][7988:7993] CHIP:DMG:        InteractionModelRevision = 1
[1657607954.931553][7988:7993] CHIP:DMG: },
[1657607954.931617][7988:7993] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0101 Command=0x0000_001A Status=0x85
[1657607954.931654][7988:7993] CHIP:TOO: Error: IM Error 0x00000585: General error: 0x85 (INVALID_COMMAND)
```
Instead, chip-tool is giving the error: `CHIP:TOO: Run command failure: src/app/data-model/Encode.h:75: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)`

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

9ac148067fb9c5a4aadab1fdecd0a8b4e22e6da0

### Platform

other

### Anything else?

## Logs

- `chip-all-clusters-app` logs: [chip-all-clusters-app.log](https://github.com/project-chip/connectedhomeip/files/12072801/chip-all-clusters-app.log)
- `chip-tool` logs: [chip-tool.log](https://github.com/project-chip/connectedhomeip/files/12072804/chip-tool.log)

## Test plan verification steps

- The test case steps and expected results were taken from: https://groups.csa-iot.org/wg/matter-csg/document/28567

## Additional testing

I also tested the same test case using `chip-lock-app`, but go the same result:

- `chip-lock-app` logs: [2_chip-lock-app.log](https://github.com/project-chip/connectedhomeip/files/12072861/2_chip-lock-app.log)
- `chip-tool` logs: [2_chip-tool.log](https://github.com/project-chip/connectedhomeip/files/12072864/2_chip-tool.log)

## Additional Comments

This bug has been found when trying to run `TC-DRLK-2.6` and `TC-DRLK-2.8` in Test Harness. This same error is also happening when running `TC-DRLK-2.6`. I believe the root cause is probably be the same.

- Logs from Test Harness running both `TC-DRLK-2.6` and `TC-DRLK-2.8`: [UI_Test_Run_2023_07_14_16_03_37.log](https://github.com/project-chip/connectedhomeip/files/12072906/UI_Test_Run_2023_07_14_16_03_37.log)

### Test Harness related issues

- `TC-DRLK-2.8` issue: https://github.com/CHIP-Specifications/chip-certification-tool/issues/897
- `TC-DRLK-2.6` issue: https://github.com/CHIP-Specifications/chip-certification-tool/issues/895

ccruzagralopes, 2023-07-17:
FYI @raju-apple @mikaelhm @krypton36

bzbarsky-apple, 2023-07-17:
> Instead, chip-tool is giving the error: CHIP:TOO: Run command failure: src/app/data-model/Encode.h:75: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)

This is a chip-tool binary that was somehow compiled incorrectly.  It's not event testing the server in this case, because it is failing to send the invalid value to start with.

`examples/chip-tool/include/CHIPProjectAppConfig.h` has this bit:

```
#define CHIP_CONFIG_IM_ENABLE_ENCODING_SENTINEL_ENUM_VALUES 1
```

but it looks like this specific chip-tool binary was compiled without that setting somehow.

@ccruzagralopes Where did this chip-tol binary come from?

mikaelhm, 2023-07-17:
@bzbarsky-apple 
The docker file for this lives in SDK.

So if it's built incorrectly, we need to fix this in here:
https://github.com/project-chip/connectedhomeip/blob/master/integrations/docker/images/chip-cert-bins/Dockerfile#L209

mikaelhm, 2023-07-17:
The `chip-tool` compilation hasn't been changed since this PR by @andy31415:
https://github.com/project-chip/connectedhomeip/pull/23874

Not sure if something changed in `chip-tool` compilation since, or it's always been wrong.

bzbarsky-apple, 2023-07-17:
I think it's always been wrong.  This bit:
```
    source scripts/activate.sh \
    && gn gen out/debug --args='chip_mdns="platform" chip_inet_config_enable_ipv4=false' \
    && ninja -C out/debug
```
does the "unified" build that ignores app-specific configs, because it's just kind of broken.  We should probably be building chip-tool as part of the `scripts/build/build_examples.py` bits around https://github.com/project-chip/connectedhomeip/blob/ba335e93f4a2866140310b309ab99b9506fdaab1/integrations/docker/images/chip-cert-bins/Dockerfile#L224C8-L224C39 instead.

mikaelhm, 2023-07-17:
I think the `scripts/build/build_examples.py` way was broken at the time. Let me see if I can find the original pull request that had this feedback.

bzbarsky-apple, 2023-07-17:
Totally believable.  Anyway, moving to building chip-tool as part of the build_examples bits (like it is in Matter CI) should resolve this problem....

mikaelhm, 2023-07-17:
We will try that!

If we add `--target linux-x64-chip-tool-ipv6only` how do we specify the `--args='chip_mdns="platform"'`

bzbarsky-apple, 2023-07-17:
> If we add `--target linux-x64-chip-tool-ipv6only` how do we specify the `--args='chip_mdns="platform"'`

@andy31415 ?

Note that you could also just do:

```
scripts/examples/gn_build_example.sh examples/chip-tool out/whartever chip_mdns ='"platform"'
```

to build just chip-tool if the `build_examples` bit is not flexible enough.

andy31415, 2023-07-18:
I believe the target supports `-platform-mdns` variant if you need platformmdns. So something like:

```
--target --target linux-x64-chip-tool-ipv6only-platform-mdns
```

may work.

andy31415, 2023-07-18:
However I am somehow unclear what platform-mdns has to do with constraint errors. I added the above describing how to set up platform-mdns, however I believe we should support both platformmdns or minmdns in the same way.

mikaelhm, 2023-07-18:
@ccruzagralopes That `build_python.sh` issue could be unrelated.

ccruzagralopes, 2023-07-18:
> @ccruzagralopes That `build_python.sh` issue could be unrelated.

I agree. I just found it odd because I had used this same Dockerfile (without the mentioned changes) yesterday and successfully built a new image, so I don't see another reason for this issue to happen now and not yesterday

bzbarsky-apple, 2023-07-19:
@ccruzagralopes I believe the virtualenv thing was a known issue on Matter tip for a day or so.  Update to tip as of right now, just in case.

Note that skipping `ninja -C out/debug` might also skip building other things (not just chip-tool) that the docker image then tries to copy.... So you might need both.

ccruzagralopes, 2023-07-19:
@bzbarsky-apple I realized that the `build_python.sh` script has been changed, so the `-i` flag that was being used in the Dockerfile now means something else and shouldn't be used in this context. When I removed that I stoped getting the virtualenv error because I don't actually need to use that.

About the other things that `ninja -C out/debug` builds, I noticed that it was `chip-shell` and `chip-cert` and that those are also possible to be built using `build_examples.py`, so I did that.

With these changes, I was able to build a new image that didn't fail `TC-DRLK-2.8`. `TC-DRLK-2.6` is still failing with a UNSUPPORTED_COMMAND error, so I'll look into the logs today to try to figure out if this error is related to the build or something else.

The Dockerfile that I used and was able to build the image was: [Dockerfile.txt](https://github.com/project-chip/connectedhomeip/files/12094935/Dockerfile.txt)
I plan to open a PR for this change once I investigate `TC-DRLK-2.6`'s failure.

bzbarsky-apple, 2023-07-19:
@ccruzagralopes Great!  TC-DRLK-2.6 failing with UNSUPPORTED_COMMAND sounds like a different issue, fwiw....  Is it running against lock-app or chip-all-clusters-app?  Because all-clusters-app does not implement all the lock bits.

ccruzagralopes, 2023-07-19:
@bzbarsky-apple I was running with chip-all-clusters-app, that was the issue 😅 When I ran it using chip-lock-app it passed

cjandhyala, 2023-08-09:
closing the issue based on your comment @ccruzagralopes , feel free to re-open if this is still an issue

closed 2023-08-09 by cjandhyala