project-chip/connectedhomeip #28006: [Workflow] How to include Matter in a project, rather than having to include a project in Matter ?

Arnaud-Le-Cossec, 2023-07-17:
### Workflow issues

Hello !

We're working on a Matter application, but we've been surprised by the way Matter is structured, or more precisely, by the way projects are built: it seems that the only way to build your project is inside the "example" folder, which seems odd. 

How are standalone applications supposed to be built (not examples)? Is there a way to include Matter in a project, rather than having to include a project in Matter?

Thanks in advance!

P.S. Is a CMake version of Matter planned? The company I work for is more used to this and I don't think it's the only one.

### Platform

all

xvortep, 2023-09-13:
Hi, do you have any updates? It seems we have the same issue here.

antonio-gabriele, 2024-01-05:
All of us have the issue. For now it's not easy for device manufacturer to give trust to. Could be matter born died?

antonio-gabriele, 2024-01-09:
You can work on this. https://github.com/antonio-gabriele/project-chip-template