project-chip/connectedhomeip #28017: [Build] Compiling imxlinux_examples.sh script for Google NDK

beyor98, 2023-07-18:
### Build issue(s)

I want to adapt the matter to my arm chip.

I am interested in connectedhomeip/scripts/examples/imxlinux_ Modify the example.sh file and replace the cross compilation tool with Google NDK tool,
The modified script content is as follows:

`#!/usr/bin/env bash

set -e
set -x

release_build=true
if [ "$3" = "debug" ]; then
    release_build=false
fi

PLATFORM_CFLAGS='-DCHIP_DEVICE_CONFIG_WIFI_STATION_IF_NAME=\"mlan0\"", "-DCHIP_DEVICE_CONFIG_LINUX_DHCPC_CMD=\"udhcpc -b -i %s \"'
gn gen --check --fail-on-unused-args --root="$1" "$2" --args="target_os=\"linux\" target_cpu=\"arm\" arm_arch=\"armv7-a\"
treat_warnings_as_errors=false
import(\"//build_overrides/build.gni\")
sysroot=\"$sdk_target_sysroot\"
target_cflags=[\"$PLATFORM_CFLAGS\"]
custom_toolchain=\"/home/w550898/zk/connectedhomeip/build/toolchain/custom\"

target_cc=\"/home/w550898/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/bin/armv7a-linux-androideabi19-clang\"
target_cxx=\"/home/w550898/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/bin/armv7a-linux-androideabi19-clang++\"
target_ar=\"/home/w550898/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/bin/llvm-ar\"
$(if [ "$release_build" = "true" ]; then echo "is_debug=false"; else echo "optimize_debug=true"; fi)"

ninja -C "$2"
`

after replacing the cross compilation Toolchain (armv7a-linux-androideabi19-clang), is compiled, and the error is reported as follows:
```
Done. Made 232 targets from 211 files in 76ms
+ ninja -C examples/lighting-app/linux/out/arm
ninja: Entering directory `examples/lighting-app/linux/out/arm'
[68/534] c++ obj/third_party/connectedhomeip/src/inet/libInetLayer.InetInterface.cpp.o
FAILED: obj/third_party/connectedhomeip/src/inet/libInetLayer.InetInterface.cpp.o 
/home/w550898/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/bin/armv7a-linux-androideabi19-clang++ -MMD -MF obj/third_party/connectedhomeip/src/inet/libInetLayer.InetInterface.cpp.o.d -Wconversion -march=armv7-a -Os -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-unused -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -DCHIP_DEVICE_CONFIG_WIFI_STATION_IF_NAME=\"mlan0\" -DCHIP_DEVICE_CONFIG_LINUX_DHCPC_CMD=\"udhcpc\ -b\ -i\ \%s\ \" -fmacro-prefix-map=../../third_party/connectedhomeip/= -std=gnu++14 -fno-rtti -Wnon-virtual-dtor -DNDEBUG -DCHIP_HAVE_CONFIG_H=1 -I../../third_party/connectedhomeip/src/include -I../../third_party/connectedhomeip/src -Igen/include -I../../third_party/connectedhomeip/zzz_generated/app-common -I../../third_party/connectedhomeip/examples/lighting-app/linux/include -I../../third_party/connectedhomeip/config/standalone -I../../third_party/connectedhomeip/third_party/nlassert/repo/include -I../../third_party/connectedhomeip/third_party/nlio/repo/include -c ../../third_party/connectedhomeip/src/inet/InetInterface.cpp -o obj/third_party/connectedhomeip/src/inet/libInetLayer.InetInterface.cpp.o
../../third_party/connectedhomeip/src/inet/InetInterface.cpp:522:9: error: use of undeclared identifier 'if_freenameindex'; did you mean 'if_nameindex'?
        if_freenameindex(mIntfArray);
        ^
../../third_party/connectedhomeip/src/inet/InetInterface.h:40:8: note: 'if_nameindex' declared here
struct if_nameindex;
       ^
../../third_party/connectedhomeip/src/inet/InetInterface.cpp:536:22: error: assigning to 'struct if_nameindex *' from incompatible type 'if_nameindex'
        mIntfArray = if_nameindex();
                     ^~~~~~~~~~~~~~
../../third_party/connectedhomeip/src/inet/InetInterface.cpp:617:9: error: use of undeclared identifier 'freeifaddrs'
        freeifaddrs(mAddrsList);
        ^
../../third_party/connectedhomeip/src/inet/InetInterface.cpp:633:23: error: use of undeclared identifier 'getifaddrs'; did you mean 'ifaddrs'?
            int res = getifaddrs(&mAddrsList);
                      ^
/home/w550898/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/bin/../sysroot/usr/include/ifaddrs.h:45:8: note: 'ifaddrs' declared here
struct ifaddrs {
       ^
../../third_party/connectedhomeip/src/inet/InetInterface.cpp:724:20: error: use of undeclared identifier 'getifaddrs'; did you mean 'ifaddrs'?
    const int rv = getifaddrs(&ifaddr);
                   ^
/home/w550898/Android/Sdk/ndk/23.2.8568313/toolchains/llvm/prebuilt/linux-x86_64/bin/../sysroot/usr/include/ifaddrs.h:45:8: note: 'ifaddrs' declared here
struct ifaddrs {
       ^
../../third_party/connectedhomeip/src/inet/InetInterface.cpp:749:5: error: use of undeclared identifier 'freeifaddrs'
    freeifaddrs(ifaddr);
    ^
6 errors generated.
[81/534] ACTION //third_party/connectedhomeip/examples/lighting-app/lighting-common:lighting-common_zapgen_zap_pregen(/home/w550898/zk/connectedhomeip/build/toolchain/custom:custom)
ninja: build stopped: subcommand failed.
```

I don't know if there are missing libraries or script configuration issues

### Platform

android

andy31415, 2023-07-20:
You seem to be customzing the clang versions, guessing for older android versions. The current Android builds require a more recent android & ndk version (we did not try to go too far back because old android versions would not have good ipv6 support anyway)

beyor98, 2023-07-21:
I use the Android SDK 26 & NDK 23.2.8568313 version,
Refer to the following https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/android_building.md.
I'm not sure if this version is suitable.

jacqueshsu, 2023-07-31:
Try it with armv7a-linux-androideabi29-clang

andy31415, 2024-10-24:
Closing as a "newer NDK/android target version is required" (you need if_freenameindex supported)

closed 2024-10-24 by andy31415