project-chip/connectedhomeip #28018: [CERT-TEST-FAILURE] TC-SMCO-2.1-Test Failed

Saravana-kr22, 2023-07-18:
### Feature Area

Area:SmokeCO2

### Test Case

TC-SMCO-2.1

### Reproduction steps

Use below command to read the Expiry-date 
`./chip-tool smokecoalarm read expiry-date  1 1`

### Bug prevalence

Whenever i do this

### GitHub hash of the SDK that was being used

b6caecf7860072bd4e55be43dc830b5ce990ddef

### Platform

raspi

### Anything else?

As the spec and test-plan the Expiry-date value should be in date format (epoch-s). But while reading the expiry-date attribute i got the value `0` as output, if this is not valid SDK needs to be updated as per the Spec and Test-Plan.

> [1689665561.653439][46473:46475] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_005C Attribute 0x0000_000C DataVersion: 969313297
[1689665561.653454][46473:46475] CHIP:TOO:   ExpiryDate: 0
[1689665561.653490][46473:46475] CHIP:EM: <<< [E:37209i S:34390 M:118386329 (Ack:231000893)] (S) Msg TX to 1:0000000000000001 [7A4E] --- Type 0000:10 (SecureChannel:StandaloneAck)

**Reference:** 
Spec:  https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/SmokeCOAlarm.adoc#expirydate-attribute
Test-Plan : https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/smco.adoc#tc-smco-2-1-attributes-with-dut_server

ericzijian1994, 2023-07-19:
Because this value is customized and the default value in the example is 0.

Survensa, 2023-07-19:
> Because this value is customized and the default value in the example is 0.

Thank you for confirmation @ericzijian1994

closed 2023-07-19 by Survensa