project-chip/connectedhomeip #28042: [BUG] chip-tool increasing repeat-count will get timeout error

resghst, 2023-07-19:
### Reproduction steps

I try Matter light's toggle command with repeat-count. The toggle command is working on a small repeat-count, but increasing repeat-count will get a timeout error.

target device board: BRD4161A (silab)

small repeat count command:
$ chip-tool onoff toggle 8 1 --repeat-count 4 --repeat-delay-ms 100
large repeat count command:
$ chip-tool onoff toggle 8 1 --repeat-count 10 --repeat-delay-ms 100

[large_repeat_count.log](https://github.com/project-chip/connectedhomeip/files/12090808/large_repeat_count.log)
[small_repeat_count.log](https://github.com/project-chip/connectedhomeip/files/12090810/small_repeat_count.log)

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

6382144be8

### Platform

other

bzbarsky-apple, 2023-07-19:
In the "small" log, what I see is that we send all the commands over the course of ~300ms, as expected.  Then we get responses to them at the 430, 430, 430, 500ms marks after the first command was sent.

In the "large" log, we send the commands over the course of ~900ms.  We get responses at the 1070ms, 1070ms, 1070ms, 1070ms, 1070ms, then nothing.

Why did the server not respond to the other command packets?  Hard to say without server logs.   Could you attach server logs, please?

resghst, 2023-07-19:
Hi @bzbarsky-apple
Here is device log
The attach server logs means otbr-agent log?

[small_repeat_count(device).log](https://github.com/project-chip/connectedhomeip/files/12091450/small_repeat_count.device.log)
[large_repeat_count(device).log](https://github.com/project-chip/connectedhomeip/files/12091451/large_repeat_count.device.log)

bzbarsky-apple, 2023-07-19:
"attach server logs" means "please attach logs to this issue that come from the Matter server (i.e. the thing the command is being sent to).

Here's the server log for the first command we fail to get a response for:
```
[00:00:51.433][info  ][EM] >>> [E:61029r S:32751 M:227153524] (S) Msg RX from 1:000000000001B669 [36AE] --- Type 0001:08 (IM:InvokeCommandRequest)
...
[00:00:51.491][error ][CSL] PacketBuffer: pool EMPTY.
[00:00:51.492][detail][DMG] Decreasing reference count for CommandHandler, remaining 0
[00:00:51.492][error ][DMG] Failed to send command response: 3
```

So the server tries to send the response, but is out of network buffers, so can't manage to do it.

What value of CHIP_IM_MAX_NUM_COMMAND_HANDLER was this server compiled with?  How big is the buffer pool it was compiled with?

resghst, 2023-07-20:
Hi @bzbarsky-apple 
The CHIP_IM_MAX_NUM_COMMAND_HANDLER used 4. (this is the default value)

bzbarsky-apple, 2023-07-21:
OK, and how big is the buffer pool?

resghst, 2023-07-21:
@bzbarsky-apple 
Following is buffer pool size setting
#define PBUF_POOL_SIZE (8)
#define CHIP_SYSTEM_CONFIG_PACKETBUFFER_POOL_SIZE 8

bzbarsky-apple, 2023-07-21:
It's definitely a device issue: the device is running out of buffers and then misbehaving. Though apparently it is managing to allocate a buffer to send an ack, just not a buffer to send a command response.

resghst, 2023-07-21:
@bzbarsky-apple 
I think this issue is a Matter stack issue. This issue will occur on all Matter devices, not only on silicon lab's board.
Maybe change the strategy or choose a good buffer pool and the number max command, I have no idea.

Thanks you.

bzbarsky-apple, 2023-07-21:
The buffer pool and command are part of the device configuration.  Devices should generally select them based on their needs....  The stack has some defaults, but those are defaults.

That said, it's possible this specific platform has bad defaults too.

bzbarsky-apple, 2023-07-21:
But just to be clear, this needs to be fixed on the device side, whether in the device config or in the Matter stack defaults.  chip-tool is not the issue.

closed 2024-07-11 by resghst