project-chip/connectedhomeip #28046: Is OpenWrt compatible?

S030602, 2023-07-19:
I am working on a project which includes the BCM63xx series SoC, which I want to access and control using OpenWrt. But I am not certain that, Is OpenWrt compatible with Matter?

AllsVoid, 2023-07-20:
Hi，have the same question as you. https://github.com/project-chip/connectedhomeip/issues/27932#issue-1802363126
From what I know so far, Matter over Wi-Fi or Thread requires specific soc support(eg: Thread needs nRF7002; Wi-Fi needs Esp32)
I made an app following [this](https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/linux) guide. 
On Ubuntu, you can only use ip connection, and when using Wi-Fi, it will prompt that the device cannot be found.

testuser7, 2023-09-29:
There is an implementation of Matter Controller Server in python: https://github.com/home-assistant-libs/python-matter-server