project-chip/connectedhomeip #28059: [CERT-TEST-FAILURE] TC-DISHALM-3.3 - Test failed

Rajashreekalmane, 2023-07-19:
### Feature Area

Area:Dishwashers

### Test Case

TC-DISHALM-3.3 

### Reproduction steps

All-clusters-app has to be enabled with below mentioned attribute :
State attribute

All-clusters-app has to be enabled with below mentioned command :
Reset

All-clusters-app has to be enabled with below mentioned event :
Notify event

system configuration
Chip-tool - RPI-4, 8GB RAM
DUT - RPI - RPI-4, 8GB RAM
Platform : onnetwork
Dut application: dishwasher-app

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

701fcb21f9a1bc4c7792c6e0709a3c74101fa6f5

### Platform

raspi

rbultman, 2023-07-19:
This is being addressed in PR #27748.

Rajashreekalmane, 2023-07-26:
implemented in latest TOT : 5b4f8004662d00bdb111367fec7d3ea978c23372

closed 2023-07-26 by Rajashreekalmane