project-chip/connectedhomeip #28101: [Tooling] Improve Test Failure Message in matter_yamltests Runner

rquidute, 2023-07-20:
### Reproduction steps / Feature

When we are running a test case in TH (eg. TC-CC-3.2) , when a test step fails, we received `log_entry.message = 'Constraints check failed'` from **matter_yamltests Runner**. We noticed that the message in `log_entry.exception.message = ‘The response value (0) should be greater or equal to the constraint but 0 < 215.'` has more details. 

Also, we noticed that depending on the failure the exception object is returned or not.

We'd like to request the `log_entry.message` to have the current message + `log_entry.exception.message.`
For example: 
`log_entry.message = 'Constraints check failed: The response value (0) should be greater or equal to the constraint but 0 < 215.'`

### Platform

python

### Type

Manually tested with SDK

### (Optional) If manually tested please explain why this is only manually tested

Testing running the **TC-CC-3.2** using the TH

### Anything else?

Debug session, with both **message** and **exception.message** populated. The exception message has more details in the failure.

[image]

bzbarsky-apple, 2023-07-20:
@vivien-apple

vivien-apple, 2023-07-21:
@rquidute I am a bit confused. What is the issue exactly ? My understanding of this is that `matter_yamltests` gives you a log entry that contains both a message and an exception. This log entry is then processed by the certification tool backend in `app/chip_tool/test_case.py`.

Is the issue that `handle_logs` adds `log_entry.message` all the time without checking if there is a better message ? https://github.com/CHIP-Specifications/chip-certification-tool-backend/blob/develop/app/chip_tool/test_case.py#L252

I am asking because what you expect from the SDK itself is unclear to me. Sorry if I misunderstood something obvious.

rquidute, 2023-07-24:
Hi @vivien-apple, thanks for your feedback. 
We'd expected that the `log_entry.message` was more specific, so we don't need to check the exception message.