project-chip/connectedhomeip #28118: [Build] Android CHIPTool failed

woody-apple, 2023-07-20:
### Build issue(s)

https://github.com/project-chip/connectedhomeip/actions/runs/5604176594/job/15182456147

woody-apple, 2023-07-20:
Another here: https://github.com/project-chip/connectedhomeip/actions/runs/5602013665/job/15175721320

woody-apple, 2023-07-20:
Another here: https://github.com/project-chip/connectedhomeip/actions/runs/5594249511/job/15153019730

woody-apple, 2023-07-20:
Another: https://github.com/project-chip/connectedhomeip/actions/runs/5594171552/job/15152788365

woody-apple, 2023-07-20:
https://github.com/project-chip/connectedhomeip/actions/runs/5594200898/job/15152878815

woody-apple, 2023-07-20:
https://github.com/project-chip/connectedhomeip/actions/runs/5594185817/job/15152834249

woody-apple, 2023-07-20:
Tagging as common failure, @andy31415 can someone take a peek?

woody-apple, 2023-07-20:
https://github.com/project-chip/connectedhomeip/actions/runs/5594171552/job/15152788365

woody-apple, 2023-07-20:
This appears to be our top build failure looking at all builds over the last 4 days

andy31415, 2023-07-20:
I am working with pigweed to remove the need of creating a mapfile, since that seems to pollute too much our output (and may be the cause of the failure because it is 800MB).

Will re-assess once we have a pigweed update with that change (expecting sometime next week, assuming we can fix other failures that I saw relating to RPC builds and tracing)

andy31415, 2023-07-21:
Newest pigweed updated in #28161 , if any new failures occur we should be able to investigate.

I assume fixed for now because we save 800MB of space and reduce log sizes. So closing for now, but can re-open if not actually fix.

closed 2023-07-21 by andy31415