project-chip/connectedhomeip #28127: [1.2] Address review comments from PR 27990

bzbarsky-apple, 2023-07-20:
### Reproduction steps

https://github.com/project-chip/connectedhomeip/pull/27990 was merged before reviews were submitted (but while it was being reviewed).  This is an issue to track addressing the review comments in https://github.com/project-chip/connectedhomeip/pull/27990#pullrequestreview-1539765953

@mideayanghui 

### Bug prevalence

N/A

### GitHub hash of the SDK that was being used

5cd5320

### Platform

core

### Type

Common Cluster Logic

mideayanghui, 2023-07-25:
@bzbarsky-apple  Boris, About this comment,
https://github.com/project-chip/connectedhomeip/pull/27990#discussion_r1269775402

I want to inherit app::Clusters::OperationalState::Events::OperationalError in  `zzz_generated/app-common/app-common/zap-generated/cluster-objects.h`  to have a generic event struct.  But there is a problem in base class.
The `GetClusterId` is a static funtion , so how to override it in generic event struct? Just like this.
`static constexpr ClusterId GetClusterId() { return Clusters::OperationalState::Id; }`

Do you have any suggestion? Thanks.

bzbarsky-apple, 2023-07-25:
`GetClusterId()` is a static function in the codegen bits, but the code calling it does not assume it's static.  It's called like so:
```
aEventData.GetClusterId()
```
via templated code where aEventData is some concrete type that we are templated over.

So I think you should be able to do something like:
```
    class GenericErrorEvent : private app::Clusters::OperationalState::Events::OperationalError::Type 
    {
        using super = app::Clusters::OperationalState::Events::OperationalError::Type;
    public:
        GenericErrorEvent(ClusterId aClusterId, const Structs::ErrorStateStruct::Type & aError) : mClusterId(aClusterId), errorState(aError) {} 
        using super::GetPriorityLevel
        using super::GetEventId;
        ClusterId GetClusterId() const { return mClusterId; }
        using super::kIsFabricScoped;
        using super::Encode;
    private:
       ClusterId mClusterId;
    };
```