project-chip/connectedhomeip #28142: [ESP32] Threading error on BLE disconnect

bzbarsky-apple, 2023-07-21:
### Reproduction steps

1. Check out revision ad5403e8e537612c2d2f903895642445d45d1e08 of Matter.
2. Build chip-tool on Mac (buggy on that revision)
3. Check out master (d0f14e0e230dc49e8ceb928232581a6b1764c39a)
4. Compile all-clusters-app for m5stack
5. Try to use any of the "ble" pairing commands from chip-tool to commissioning the m5stack.

Almost every time, when the chip-tool tears down the BLE connection (because it's buggy), the following stack is hit:

```
0x40097d62: abort at tools/esp-idf/components/newlib/abort.c:46

0x4013addf: chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:508
 (inlined by) chipDie at connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:518
 (inlined by) chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/platform/LockTracker.cpp:36

0x4013be31: chip::System::LayerImplFreeRTOS::CancelTimer(void (*)(chip::System::Layer*, void*), void*) at connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/system/SystemLayerImplFreeRTOS.cpp:101

0x401569bd: chip::Ble::BLEEndPoint::StopReceiveConnectionTimer() at connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/ble/BLEEndPoint.cpp:1487

0x40156c64: chip::Ble::BLEEndPoint::DoClose(unsigned char, chip::ChipError) at connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/ble/BLEEndPoint.cpp:331

0x40133521: chip::Ble::BleLayer::HandleConnectionError(unsigned short, chip::ChipError) at connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/ble/BleLayer.cpp:745

0x40142f97: chip::DeviceLayer::Internal::BLEManagerImpl::HandleGAPDisconnect(ble_gap_event*) at connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:1287

0x40143113: chip::DeviceLayer::Internal::BLEManagerImpl::ble_svr_gap_event(ble_gap_event*, void*) at connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/platform/ESP32/nimble/BLEManagerImpl.cpp:1370

0x40109b7d: ble_gap_call_event_cb at tools/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_gap.c:716
```

### Platform

esp32

### Type

Manually tested with SDK

bzbarsky-apple, 2023-07-21:
@shubhamdp @dhrishi

shubhamdp, 2023-07-21:
@bzbarsky-apple Will take a look!

PR #29180 (2023-11-06): [ESP32] Fix the threading issue in nimble
#### Problem
When handling BLE disconnection, `CancelTimer` is being called from NimBLE thread, causing the assert described in #28142.

#### Change Overview
Scheduling the code which causes assert on the chip stack.
Fixes #28142

#### Tests
Commissioned all-cluster-app for target C3 and ESP32-M5Stack, using Linux/Mac chip-tool. Device did not crash.