project-chip/connectedhomeip #28148: [1.2] ReplacemenProductListManager is assuming static memory allocation

bzbarsky-apple, 2023-07-21:
### Reproduction steps

The API looks like this:
```
    virtual CHIP_ERROR Next(Attributes::ReplacementProductStruct::Type & item) = 0;
```

But that struct has a CharSpan inside, and how does the callee know when it's safe to free the memory the CharSpan is pointing to?  It doesn't.

We need to stop using "static stick it all in memory" implementations to back our delegate APIs, unless we really expect all applications to do it that way....

@cliffamzn @tcarmelveilleux @chrisdecenzo 

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

d2d4701

### Platform

core

### Type

Common Cluster Logic

### Anything else?

@cliffamzn I suggest doing one of two things.  Either have separate getters for the separate parts of the struct, or have a struct type that _includes storage_ so that you can put one on the stack and the callee can copy the data into it.  The latter is what I had been suggesting you do on Slack, if you recall: inherit from one of the generated structs, pull in any method definitions, adapt your subclass for your needs...

bzbarsky-apple, 2023-07-21:
@cliffamzn Also, see https://github.com/project-chip/connectedhomeip/pull/28095#pullrequestreview-1540409006

cliffamzn, 2023-07-21:
@bzbarsky-apple - included suggestions from previous changes in the latest PR as well.

closed 2023-07-27 by cliffamzn