project-chip/connectedhomeip #28155: [Feature] replace the use of EMBER_AF_PLUGIN_MODE_BASE with a BUILD.gn variable

hicklin, 2023-07-21:
### Feature description

Remove `#define EMBER_AF_PLUGIN_MODE_BASE` form the all-clusters-app `CHIPProjectAppConfig.h` and find a solution using the BUILD.gn.

Update the Mode Base readme.

### Platform

all

hicklin, 2023-07-21:
See the conversation [here](https://github.com/project-chip/connectedhomeip/pull/27504/files/088e22c85a2ec01b9c4bfd7ffea16850748a8c94#r1243731838).

bzbarsky-apple, 2023-07-21:
Better conversation link that will not break: https://github.com/project-chip/connectedhomeip/pull/27504/#discussion_r1243731838

hicklin, 2023-07-25:
Should I close this issue given that there are no strong opinions against the current implementation? 

If matter continues to utilise zap and generated code, the best way that I think this can be implemented is to have the code regen script define `EMBER_AF_PLUGIN_MODE_BASE` with all the others in `gen_config.h` once, when at least one ModeBase derived cluster is enabled.

closed 2023-07-27 by hicklin