project-chip/connectedhomeip #28166: dynamic endpoint support - missing bits / ongoing efforts?

plan44, 2023-07-21:
`EMBER_AF_POWER_SOURCE_CLUSTER_SERVER_ENDPOINT_COUNT + CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT` but you have to include the ember config header for that.  Might be better to put that in the .cpp not here.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/28110#discussion_r1270103331_

bzbarsky-apple, 2023-07-21:
>  This can amount to significant waste of RAM.

Yes, the right way to deal with this is to move toward representing each cluster instance (i.e. pair of endpoint and cluster id) as an object, which has a pointer to its delegate directly...

> With the current direct mapping of endpointIDs into these CHIP_DEVICE_CONFIG_DYNAMIC_ENDPOINT_COUNT sized table

This is not at all what's going on.  The index into the table is not the endpoint id.  The index is determined from the endpoint id by looking at the actual list of defined endpoints.  The only place that assumes endpoint id == endpoint index is fixed endpoints.

So this part is actually fine.

> here's no way to have gaps in the dynamic endpoint list except instantiating them and then set them disabled

I am not sure what you mean.  When you instantiate a dynamic endpoint, you specify its endpoint id.  This is not related to its index in the dynamic endpoint list (and in fact they can be in the list in an order that does not match id order).

> there's no elegant way to use the attribute storage architecture which is available for fixed clusters

Correct.

> Are there existing plans or maybe even not yet published work how to evolve the SDK towards better dynamic endpoint support?

Apart from the "move toward having object-per-cluster, more or less deprecate the existing attribute store and the ZAP dependency"?  Not anything concrete....

plan44, 2023-07-22:
> @bzbarsky-apple:  This is not at all what's going on. The index into the table is not the endpoint id. The index is determined from the endpoint id by looking at the actual list of defined endpoints. The only place that assumes endpoint id == endpoint index is fixed endpoints.

Sorry for that. Indeed I had settled on the false assumption (no idea from where, probably an early bridge example) that endpoint id was just the index, offset by the number of fixed endpoints. But looking at attribute-storage.cpp now, clearly there's always been a mapping step between the two. My bad.

> I am not sure what you mean. When you instantiate a dynamic endpoint, you specify its endpoint id. This is not related to its index in the dynamic endpoint list (and in fact they can be in the list in an order that does not match id order).

This was also based on the false assumption on endpoint ids tied to indexes directly. So: not an issue at all, in reality.

>> Are there existing plans or maybe even not yet published work how to evolve the SDK towards better dynamic endpoint support?
> Apart from the "move toward having object-per-cluster

Object-per-cluster would certainly be great. Would you agree it should be done in a way transparent to cluster implementation? What I currently see in window-covering-server seems a bit odd to me, there shouldn't be any dependencies on counts of fixed/dynamic clusters there, no?

> [...] more or less deprecate the existing attribute store

I'd rather would want to extend it, similar to the way you did for the DataVersion, such that `RAM` attributes would work without app side code, and for `NVM` ones the app would only have to persist/restore opaque blocks of memory.

> [...] and the ZAP dependency"?

It would be super useful if dynamic clusters could use ZAP-defined endpoints and clusters as templates, rather than programmatically re-construct them using the `DECLARE_DYNAMIC_xxx` macros.

> [...] Not anything concrete....

So give me some time to understand `attribute-storage` in detail (and maybe clear out other false working assumptions on my part, by the way) and then maybe come up with a suggestion.

bzbarsky-apple, 2023-07-25:
> Would you agree it should be done in a way transparent to cluster implementation?

I think generally we need to rethink where the boundary between generated and non-generated bits lies.  Which may not be the right answer to your question, but I am not 100% sure what the question is asking.

> I'd rather would want to extend it

The existing attribute store uses a statically allocated slab of RAM; it just cannot handle dynamic endpoints.

>  It would be super useful if dynamic clusters could use ZAP-defined endpoints and clusters as templates,

What does "zap-defined" gain us there?

Put another way, as long as the SDK ends up auto-generating the right metadata based on how the cluster object is constructed, do we care whether it's matching the data structures ZAP spits out right now?

plan44, 2023-07-28:
> > Would you agree it should be done in a way transparent to cluster implementation?
> 
> I think generally we need to rethink where the boundary between generated and non-generated bits lies. Which may not be the right answer to your question, but I am not 100% sure what the question is asking.

I think the fact that there can be static and dynamic endpoints, and how these are implemented, is not something that cluster implementation should have to know. My question was if that statement is correct from a should-be point of view.

Looking at current window covering server, there's some struggling with a delegate table that would allow having a separate delegate for each *instance*, but then a GetDelegate function which rejects any dynamic endpoints. That's probably just a bug, but to fix it I'd get a better idea how this is *meant* to work.

I have the impression that a cluster implementation should *not* try to handle per-instance storage for state.

- "Official" cluster state is in the matter attributes, how these are implemented is none of the cluster's business. EXTERNAL type attributes serve as link to the actual hardware.
- Where external attributes are not sufficient for controlling the hardware, I think delegate *functions*, to be implemented by the app, taking an endpointId argument, would be the better solution than a per-instance delegate object, because the latter again needs one of those maximum-dynamic-endpoint sized tables. Its none of the cluster implementation's business to guess whether the hardware implementation needs extra state at all, and if so, where it is best located (maybe not in the delegate object!).
- When the cluster implementation needs per-instance private state, such as for example level-control does, this currently requires another maximum-dynamic-endpoint sized table. Getting around this would require some generic per-instance storage management that (similar to attribute storage) that *internally* handles static vs. dynamic allocation.

Which leads to:

> The existing attribute store uses a statically allocated slab of RAM; it just cannot handle dynamic endpoints.

Until now :wink: - see draft PR #28372. After living a day in `attribute-storage.cpp`, it was relatively straightforward. I hope I did not totally miss something, but at least my bridge seems to work fine using this.

bzbarsky-apple, 2023-07-28:
> My question was if that statement is correct from a should-be point of view.

Ideally, I agree.

>  That's probably just a bug

It is.

> "Official" cluster state is in the matter attributes, how these are implemented is none of the cluster's business. EXTERNAL type attributes serve as link to the actual hardware.

No, official cluster state is "somewhere", where is up to the cluster.  The attributes are a view onto that state, and for a lot of clusters not all the state is captured by attributes.

In MVC terms, the state is the model, the attributes are one view.

>  I think delegate functions, to be implemented by the app, taking an endpointId argument, would be the better solution than a per-instance delegate object, because the latter again needs one of those maximum-dynamic-endpoint sized tables

Only if the cluster itself is implemented as a bunch of global callbacks to start with.  If we have object-per-cluster-instance, then it can easily have a single pointer to its one delegate, etc.  That's how newer cluster implementations are being done, mostly.

plan44, 2023-07-28:
> In MVC terms, the state is the model, the attributes are one view.

Yes, but many clusters are implemented to only use that attribute view to the model.

With "official" cluster state I meant to say that this is the state "visible" in the specs via the data model. Of course, specs saying "output should smoothly transition within x seconds" also implies state, but rather ephemeral one.

> [...] That's how newer cluster implementations are being done, mostly.

Good to know.

What is the main motivation for that? Having a single place for the non-attribute-exposed state plus the needed callbacks into the app?

However, without cluster state in the delegate, the bunch-of-callbacks is more flexible in what the hardware side implementation can look like. The callback can look up a delegate if it needs to, but is not forced to use one avoiding that max-endpoint-count sized table.

I'm again looking at Window Covering, which seems to access state only via attributes, and having a delegate without any state and only two callbacks.

bzbarsky-apple, 2023-07-28:
> What is the main motivation for that? 

Making it easier to do dynamic composition, plus making state management simpler.

Window Covering is very much not using the "object per cluster instance" setup.

plan44, 2023-08-06:
The discussion has resolved the original questions as follows:
- endpointID to endpointIndex mapping is fine, thinking that there was an 1:1 correspondence between the two was a misconception on my part.
- the bug in window-covering-server.cpp not working with dynamic endpoints was fixed in 92287b8.
- newer cluster implementations tend to use a delegate object for each cluster *instantiation*, which can gather all needed callbacks plus internal state. Window Covering is not yet a very good example for actually using the potential of a delegate object.
- To make use of automatic attribute storage and ZAP based configuration also for dynamic endpoints, see PR #28372.

closed 2023-08-06 by plan44