project-chip/connectedhomeip #28181: [TC-DGGEN-2.1] YAML script does not follow test plan and does not validate values as intended by plan

tcarmelveilleux, 2023-07-21:
### Feature Area

Other

### Test Case

TC-DGGEN-2.1

### Reproduction steps

- Run test against a device that never updates or persists any of the following attributes
  - DGGEN.S.A0001(RebootCount)
  - DGGEN.S.A0002(Uptime)
  - DGGEN.S.A0003(TotalOperationalHours)
- Test never saves or checks whether the attributes have incremented from last boot, even though the device reboots.

Steps 3, 4 and 5 of the test plan indicate the value must change. The YAML file (src/app/tests/suites/certification/Test_TC_DGGEN_2_1.yaml) proceeds with reboot but never causes any check of the value to be requested, and does not read the values after reboot to allow logs or UI to show a change.

### Bug prevalence

100%

### GitHub hash of the SDK that was being used

2c06e6c0d545c2d0a8561f38742af2d22f5c5182

### Platform

core

manjunath-grl, 2023-08-22:
@tcarmelveilleux Modified the test TC-DGGEN-2.1 as per test-plan can you please take a look at it.
Diff link: https://github.com/manjunath-grl/connectedhomeip/blob/bffdc022e19a378e73bca529e2c39c159af827f1/src/app/tests/suites/certification/Test_TC_DGGEN_2_1.yaml

cjandhyala, 2023-08-25:
@tcarmelveilleux Could you pls take a look at the above comment ?

woody-apple, 2023-09-19:
@tcarmelveilleux Marking this as closed based on the above comment, please re-open if this is not resolved